ToFuProject/tofu #369: pip install from source on IRFM Intra failed due to missing _updateversion.py

Didou09, 2020-03-12:
Installation of tofu 1.4.3 was attempted on IRFM cluster, in a sandbox conda environment:
```
module load anaconda/python37
conda info --envs
conda create --help
conda create -n tofu-test python=3.7 pip
conda info --envs
conda init
conda activate tofu-test
```

The error message (see below) suggests that _updateversion.py was not included in the source package:

```
(tofu-test) [root@spica ~]#
(tofu-test) [root@spica ~]# pip install tofu
Collecting tofu
  Downloading tofu-1.4.3.tar.gz (480 kB)
     |████████████████████████████████| 480 kB 1.9 MB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 1:
   command: /Applications/Anaconda/python37/envs/tofu-test/bin/python /Applications/Anaconda/python37/envs/tofu-test/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpakr1wkem
       cwd: /tmp/pip-install-7gpsil44/tofu
  Complete output (18 lines):
  Traceback (most recent call last):
    File "/Applications/Anaconda/python37/envs/tofu-test/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 257, in <module>
      main()
    File "/Applications/Anaconda/python37/envs/tofu-test/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 240, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/Applications/Anaconda/python37/envs/tofu-test/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 91, in get_requires_for_build_wheel
      return hook(config_settings)
    File "/tmp/pip-build-env-yx7q6cx3/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 148, in get_requires_for_build_wheel
      config_settings, requirements=['wheel'])
    File "/tmp/pip-build-env-yx7q6cx3/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 128, in _get_build_requires
      self.run_setup()
    File "/tmp/pip-build-env-yx7q6cx3/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 250, in run_setup
      self).run_setup(setup_script=setup_script)
    File "/tmp/pip-build-env-yx7q6cx3/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 143, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 20, in <module>
      import _updateversion as up
  ModuleNotFoundError: No module named '_updateversion'
  ----------------------------------------
ERROR: Command errored out with exit status 1: /Applications/Anaconda/python37/envs/tofu-test/bin/python /Applications/Anaconda/python37/envs/tofu-test/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpakr1wkem Check the logs for full command output.
```

lasofivec, 2020-03-12:
Could you try with pip ? (just to see if there is the same behavior).
I think this could be related to #292 
which I (thought I) solved with:

https://github.com/ToFuProject/tofu/pull/348#issue-372046999

> 
> Issue #292 : the file _updateversion.py was no longer being deployed with pip:
> #292 (comment)
> I used the second solution: I took out out build-backend = "setuptools.build_meta" in pyproject.toml

Didou09, 2020-03-12:
This was already with pip, with admin rights, do you want me to do it again on my session ?

Didou09, 2020-03-30:
Fixed in devel, by PR #370

closed 2020-04-07 by Didou09