ToFuProject/tofu #382: putting tofu on conda-forge?

flothesof, 2020-05-05:
Dear @Didou09 @lasofivec,
I've recently been involved in the task of adding conda-forge builds for win, linux and osx for a package called pygalmesh, which was until then mostly a linux package. 
Long story short, I managed, with the help of the principal developer, to get conda-forge builds going in a short amount of time thanks to their fantastic support and docs (see the PR here to get a feeling for the process: https://github.com/conda-forge/staged-recipes/pull/11457).

In a nutshell, getting into conda-forge would allow easy install for all platforms, which now seems to be the case for pygalmesh. 

It involves writing a cross-platform meta.yaml recipe (which would look similar but still quite different from the one already written for tofu) based on the pypi uploaded source distribution. conda-forge takes care of the build through their CI system, iterating until the build passes on all platforms. 

Based on my experience doing this for pygalmesh, should I try and open a pull-request for getting tofu on conda-forge and report back here? 

Cheers
Florian

Didou09, 2020-05-05:
Yes ! can you create a dedicated branch ?

@lasofivec can you keep an eye on it and if necessary provide some help ?

Thanks !

flothesof, 2020-05-05:
Actually, the way conda-forge works is that you submit a PR to their staged-recipes repo. This is what I've just done here: https://github.com/conda-forge/staged-recipes/pull/11497

If the recipe works (CI etc.) it will eventually get merged to a so-called feedstock which will then be updated based on future version of tofu.

I put all of us three as potential maintainers to the tofu recipe.

flothesof, 2020-05-06:
First round of feedback: all three builds failed. See https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=155677&view=results.

As far as I can tell, all for different reasons:

- linux while running check_for_openmp in the setup
- osx due to a call to clang++ which wasn't found (although calls to clang work)
- windows due to the fact that the license.txt file is not bundled in the pypi sdist and the recipe declares its existence

I'll try to have a second round of feedback with some patching going today.

flothesof, 2020-05-06:
Current status:
- builds almost compiling on all three platforms
- not passing tests yet

I have investigated the use of openmp and as far as I can tell, openmp is only supported for linux on conda-forge (I might be wrong here, I'm planning on getting some feedback from conda-forge admins once the tests are passing).

This could work. Let's see how it goes in the next day or two...

flothesof, 2020-05-07:
Current status:
- builds ok on linux (with openmp), osx (no openmp), windows (no openmp)
- tests failing due to #377 
- license missing #387 

If say, there was a new release of tofu for 1.4.5 with #377 and #387 fixed, we could have a working conda package as far as I can tell. I'll try to patch in the meantime and see if the CI works.

Didou09, 2020-05-07:
Thanks !
Taking care of #387 today

tofu 1.4.5 on the way, as soon as @lasofivec is back on tracks (we need to review 2 PRs together before merging)

flothesof, 2020-05-11:
Hi @Didou09 

Do you have any plans to do a release that includes the license file this week so I can continue working on this issue?

Cheers,
Florian

Didou09, 2020-05-11:
Hi @flothesof , I'll do it soon this week, but 2 more small PR to be included first, in particular a nice contributor just told me he could provide an approximate geometry for DEMO (prototypal european reactor after ITER) and for TOMAS (small toroidal device used for heating antenna tests)   :-)

Didou09, 2020-05-12:
So....   tofu 1.4.5 has just been released and it includes the changes you needed:
* license is in the sdist
* all ressources in geom/input are in the sdist

Do you think you can try to go further ?

Also, I can't build the new doc because on CEA servers I can't install all the necessary sphinx dependency, can you take care of it ?
There is tuto [here](https://tofuproject.github.io/tofu/dev_tutos.html#devtutos).

The branch new-doc is up to date, all you need is, presumably, to build the html and push to branch git-pages,
Thanks !

flothesof, 2020-05-13:
I've just updated the recipe. Tests are failing.

Could you check the build log and tell me what you think? 

https://dev.azure.com/conda-forge/84710dde-1620-425b-80d0-4cf5baca359d/_apis/build/builds/160819/logs/34

Didou09, 2020-05-13:
Hum... looks like there may be data missing for tests in the sdist...
Strange because all data in included in the setup.py.
But in the light of our recent work, I guess it's because we forgot to include it in MANIFEST.in too.
I'll try to fix it today

Thanks !

flothesof, 2020-05-13:
I'm able to replicate the failures locally.

Steps:  
- dowload the archive for tofu 1.4.6 from pypi
- extract to `/temp`
- run `python setup.py build_ext --inplace` in the right folder
- run `python tofu\tests\tests01_geom\tests03_core.py`

I get the same error message. It happens during the set_dconfig routine while checking for in structures.

Didou09, 2020-05-13:
Hum, interesting, it rings a bell:
When I designed the unit tests, I also wanted to test the plotting routines to make sure they would not crash.
But they did crash on Travis because travis did not have the Qt5Agg backend, so I forced agg backend instead for unit tests to make sure they would pass on Travis.

But that should happen in unit tests only, not when you import the library for regular use...

Did you run tests just before importing ?

flothesof, 2020-05-18:
Hi @Didou09 
have you been able to take a look at the replication I posted in https://github.com/ToFuProject/tofu/issues/382#issuecomment-627837434 ? If so, do you have the same test failures?

Thanks for your feedback.
Florian

lasofivec, 2020-06-03:
Regarding the issue in https://github.com/ToFuProject/tofu/issues/382#issuecomment-627837434
It reminds me to what I found when modifying the way travis builds tofu (see #347)
The error messages have been updated since then, but I couldn't make it work by building tofu inplace, which probably means there's a linking problem. I will look into it (I wasn't able to recreate it before)

flothesof, 2020-06-03:
Thanks for looking into this! Did you manage to replicate my minimal example? Looking forward to your findings so I can progress on the conda-forge setup :)

lasofivec, 2020-06-04:
Yes, The PR #398 should fix it.
how does it work with the conda-forge ? Do we have to do a release for the tests to run again ?

flothesof, 2020-06-05:
> Yes, The PR #398 should fix it.
> how does it work with the conda-forge ? Do we have to do a release for the tests to run again ?

1. we have to do a new release
2. we have to update the PR for a feedstock on conda-forge <https://github.com/conda-forge/staged-recipes/pull/11497> with the newly uploaded sdist on pypi (I would do that)
3. wait for the conda-forge CI to tell us if it works

lasofivec, 2020-06-05:
@Didou09 maybe we can do a release on Monday? Or if you want I can do it on Sunday :)

Didou09, 2020-06-05:
I suggest we review your PR together on Monday and do a release just afterwards, ok for you ?

Didou09, 2020-10-19:
Fixed by PR #410 and available since tofu 1.4.7

closed 2020-10-19 by Didou09