ToFuProject/tofu #386: tofu pypi build does not ship the configuration files?

flothesof, 2020-05-07:
Hi,

in the context of the work I'm doing on #382 I've realized what I think is a problem. tofu is supposed, when installed by a user, to ship default configurations of machines, right?
More specifically, the sdist over at pypi (https://files.pythonhosted.org/packages/00/7b/4eb998cec2acf5df56e08a452dd7db194b63300120273a77e9f6179b9e75/tofu-1.4.4.tar.gz) when inspected only contains the deprecated geom files in tofu/geom/inputs:
[image]
This is one of the reasons the tests of the current conda-forge recipe fail: they are unable to find the configuration files.

Could someone please double check this?

Regards,
Florian

Didou09, 2020-05-07:
You're right, the first issue was spotted in #377 and fixed by PR #378 (not ncluded in 1.4.4, will be included in 1.4.5)

The second one however (test_data).... I need to double-check (might take a few days due to workload)

Thanks !

flothesof, 2020-05-07:
Thanks for your feedback. This is then a duplicate issue. I'm not sure the test_data thing is relevant. Feel free to close the issue.

Didou09, 2020-05-07:
Double check: test_data was a temporary artefact, specific to a branch that does not exists anymore.

Since the main issue is already fixed in the next release by PR #378, I'm closing the issue

closed 2020-05-07 by Didou09