ToFuProject/tofu #397: tofu messes with matplotlib backend (while generating doc, but not only then)

flothesof, 2020-05-13:
Hi,

I've been trying to generate the tofu doc under windows.
However, the tutorial thumbnails do not get generated correctly: 
[image]

I've been trying to investigate why this is happening and have been able to track down what I think is a related issue. This is a minimal example:
```
import matplotlib
matplotlib.use('Qt5Agg')
print(f"matplotlib is currently using the following backend {matplotlib.rcParams['backend']}")
import tofu as tf
print(f"matplotlib is currently using the following backend {matplotlib.rcParams['backend']}")
```
This is the output:
```
matplotlib is currently using the following backend Qt5Agg
matplotlib is currently using the following backend agg
```

Why is tofu changing the backend? From a user perspective, it's surprising since this can have a lot of side effects.

A bad one in my case is when I run the tutorial standalone files. No interactive plots show up at all, unless I fix this by first setting the matplotlib backend in my interactive session and then importing tofu.

Didou09, 2020-05-13:
It's because tofu generates interactive figures, which are only compatible with interactive backends, so tofu forces the use of Qt5Agg .
What I don't understand is that in your example it looks like it is forcing 'agg', which is not interactive and does not correspond to what I remember coding....
I need to check this up, thanks for detecting this !

flothesof, 2020-05-13:
Thanks for the details. 
As a side question, what does the above code produce as output on your system?

flothesof, 2020-05-13:
Just doing a quick search using github, I found this in `tofu/__init__.py`:

```Python
# For tests without display with nosetests
if 'matplotlib.pyplot' not in sys.modules.keys():
    import matplotlib
    matplotlib.use('agg')
    del matplotlib
```

The problem with this is that it means that unless someone has already loaded pyplot, it will default to 'agg'.
A better way I can think of the top my head, because I've encountered it before (but with pytest), is to set the backend in the `setup.cfg` along with other common nosetests options. I'm not sure whether that's possible, but another good way is described in this [blog post](https://www.davidketcheson.info/2015/01/13/using_matplotlib_image_comparison.html) and involves running the Travis nose tests through a `tests.py` file that first sets the backend.

lasofivec, 2020-06-03:
This looks like a duplicate of #216 
@flothesof how did you solve this with pytest ? since we are going to move on from nose to pytest soon.
At the beginning, I was a fan of modifying the `setup.cfg` however now I don't find it so practical: when you need to change a configuration element, you will have to make a local change to the file, but then you have to be careful not to commit/push it as the changes are only useful to you.

flothesof, 2020-06-03:
Hi @lasofivec 
nice to see you're back :)
I've lookied into your question and actually, the problem I talked about was quite similar to the one here: I was using `nose` to test my library and didn't want the figures to block the tests. 
Hence, I used a `run_test.bat` file which did something like this:

```
python -c "import matplotlib; matplotlib.use('agg'); import os; import nose; nose.main()" -v packagename
```
Instead of using a global hack, this just patches the way the tests are run. 

Regarding what you're saying about setup.cfg, I'm thinking that this should be a global setting: tests would always be run with the agg backend, just like it is today. But the backend setting would only happen when running the tests, not in tofu itself as it is today.

Cheers
Florian

lasofivec, 2020-06-04:
The thing is we run the tests in different scenarios:
- First of all in Travis, of course the solution we adopt has to be compatible with this (which yours do)
Locally, to see if we haven't broke anything on the library:
- by running `nosetests` locally
- by running a single test...

While answering this, I realized that we set the backend to `agg` manually on one test:
https://github.com/ToFuProject/tofu/blob/96fa28ccc1777936c00e9b7fd1635aad081b7586/tofu/tests/tests01_geom/tests01_GG.py#L10

Do you think if we did this for all tests it would solve the issue ?

flothesof, 2020-06-05:
To answer your question: I think setting the 'agg' backend for all tests would solve the issue. 

This would preferably be done in one place (like in my past example) instead of a test by test fashion (like the one you quoted).

The benefits would be:
- works on Travis
- doesn't interfere with the documentation like it does now
- does what @Didou09 thought should is already done but isn't "It's because tofu generates interactive figures, which are only compatible with interactive backends, so *tofu forces the use of Qt5Agg*."

Possible downsides:
- if you run a single test and want to see the output, you have to do something manually (not necessarily though, since if you just configure the main testing utility to use a backend, it could be transparent for individual tests)

So all in all, I believe there's a strong case for removing this:
```python
# For tests without display with nosetests
if 'matplotlib.pyplot' not in sys.modules.keys():
    import matplotlib
    matplotlib.use('agg')
    del matplotlib
```

lasofivec, 2020-06-05:
For the individual testing you still don't want to see the plots. The examples and the gallery in the doc are the only way we verify the outputs are untouched. The unit tests test against hard-coded data and cross-validates functions.