ToFuProject/tofu #400: Adding COMPASS geometry

Didou09, 2020-06-05:
The Czech plasma research institute recently made a lot of info available online about their tokamak [COMPASS](http://www.ipp.cas.cz/vedecka_struktura_ufp/tokamak/).
It would be nice get the geometry and add it to the next tofu release.

Do you think you could take care of this @flothesof ?

flothesof, 2020-06-05:
Sure, I'll look into it next week!

Cheers,
Florian

flothesof, 2020-06-09:
Hi @Didou09,
I've been looking into this and the available information isn't that great, unfortunately.
The best poloidal plane drawing I could find is this one:
[image]
Taken from the conceptual design paper about Compass-U https://www.sciencedirect.com/science/article/pii/S0920379617302053#fig0030

Do you think the bold structures are enough to model Compass-U in tofu? As you may notice they're not closed, which seems surprising to me.

If you're positive this is enough, I can apply some image analysis and extract the bold surfaces, close them, find the scale and do a PR for tofu.

What do you think?

Cheers
Florian

flothesof, 2020-06-09:
After further investigation, I think the best illustration is [this one](http://www.ipp.cas.cz/miranda2/export/sitesavcr/ufp/sys/galerie-obrazky/Single_null_double_null.png): 
[image]
I could extract the different geometrical elements out of this structure.

flothesof, 2020-06-09:
Yet another image, with scale (yay!) [here](http://www.ipp.cas.cz/miranda2/export/sitesavcr/ufp/sys/galerie-obrazky/single_double_conf.png):

[image]

Didou09, 2020-06-09:
Yes !
Cool, if you can extract the elements from this scaled sketch, it would be cool!
Also, tofu needs at least one closed surface, maybe you could use this scaled sketch to extract the scaled coordinates of the first figure (of which you would closed the thick light blue elements by linking them together as a single contour, and leave the dark blue and green elements as individual thick elements) ?

Didou09, 2020-06-22:
Hey @flothesof ,
Any chance you could try extracting a nice geometry for COMPASS ?
The first figure you suggested looks a nice basis, using the scale provided in the last figure.
I'll try to release a new version soon to accelerate conda-forge integration, and it would be nice to include COMPASS in it ;-)

closed 2020-10-19 by Didou09