ToFuProject/tofu #409: Bug running tests/tests01_geom/tests03_core.py

flothesof, 2020-10-10:
This is a specific bug report for what seems to be the only blocking issue before being able to be on condo-forge (see #382).

How to reproduce:

- install tofu from tofu-1.4.6.tar.gz with `pip install tofu-1.4.6.tar.gz`
- extract tofu/tests/* to independent folder named `tofu-tests_only`
- run `python tofu-tests_only/tofu/tests/tests01_geom/tests03_core.py`

I expect this script to run without any problems. But it seems it isn't. 

My current analysis is that something is messed up in the way the two configs 'Lin' and 'Tor' are built: lS is always empty which triggers a test error further down the line.

```
#######################################################
#
#  Creating Config objects and testing methods
#
#######################################################
dconf = {}
for typ in dobj.keys():
    lS = []
    for c in dobj[typ].keys():
        lS += list(dobj[typ][c].values())
    Lim = None if typ=='Tor' else [0.,10.]
    dconf[typ] = tfg.Config(Name='Test%s'%typ, Exp=_Exp,
                            lStruct=lS, Lim=Lim,
                            Type=typ, SavePath=_here)
```

As a side note, this test file is incomprehensible to me (lots of commented stuff everywhere, no outline of what the purpose of the tests is...).

Any thoughts @Didou09 @lasofivec? Can any of you try to reproduce locally?

Regards,
Florian

Didou09, 2020-10-11:
OK, In reproduced it, thanks for the detailed procedure  :-)
Working on it today

Didou09, 2020-10-11:
Ok, the problem is coming from missing test files (.txt) in tests01_geom/tests03_core_data/

The tests try top load these files into a list, so the list is empty and of course tofu later identifies that there is some missing data...

Didou09, 2020-10-11:
There was a typo in the setup.py that was preventing the proper directory to be included in the sdist package...

Didou09, 2020-10-11:
Correcting the typo fixes it.
The last tests (tests09_tutorials) does not pass because they need access to the doc however, which is not included when we copy the tests alone.
They do pass on the installed tofu package, so it should be fine

flothesof, 2020-10-11:
As I said in the PR, I have confirmed that this fixes the issue.

Now, a new release is needed (so that pypi has the new source) and so that I can re-generate a conda-forge build with it. 

@Didou09 can you do a new release this week?

Didou09, 2020-10-11:
Yes :-)

closed 2020-10-19 by Didou09