ToFuProject/tofu #440: Updating our CI strategy

lasofivec, 2020-12-11:
So since I have been away it seems it has become more urgent to update our whole CI approach. 
First of all because Travis is really slow lately. I was looking into adding Windows as an OS in the `travis.yml` but for one push I had to wait 2h in the queue... This was also a (minor?) problem for the latest releases. I don't know what you think but I think it is extremely frustrating to say the least.

It seems that many open source projects are migrating away from Travis. One of the first alternatives I have seen is Github's Actions. I want to read a little bit more into it to see if we can and how easy it is. But, since there are many projects doing this it's well documented:
https://docs.github.com/en/free-pro-team@latest/actions/learn-github-actions/migrating-from-travis-ci-to-github-actions

Which brings me to my second point:
We probably should finally move from `nose` to `pytest`. We talked about it during our meeting last year and in issue #397

I would suggest trying to do this two things during the rest of the month of december. So we start 2021 with a clean slate.

So potentially closing this issue would close issues: #397 #232 #295 #397 ... and I'm probably missing some

lasofivec, 2020-12-11:
Some resources on Travis slowness and people migrating out of it (thanks @flothesof for the first two)

https://blog.travis-ci.com/2020-11-02-travis-ci-new-billing
https://twitter.com/alex_gaynor/status/1336098972864303104
https://github.com/josdejong/mathjs/issues/2024

flothesof, 2020-12-15:
Adding some reading about Github Actions testing with Python: https://docs.github.com/en/free-pro-team@latest/actions/guides/building-and-testing-python

lasofivec, 2020-12-16:
Yes.
If I understood correctly Travis slowness should be resolved by January, but I still think it's worth while to have a backup plan.
I was looking to other FOSS and many use either: travis, Github's Actions, Azure or a combination of those.