ToFuProject/tofu #447: Create unit tests by comparing images

lasofivec, 2020-12-17:
Since now we switched (or we are on the verge of switching) to `pyplot` we can use `pyploy-mpl` to compare resulting images:

https://github.com/matplotlib/pytest-mpl

Didou09, 2020-12-18:
> Since now we switched (or we are on the verge of switching) to `pyplot`

You mean `pytest `right ?

Didou09, 2020-12-18:
Good idea !

lasofivec, 2020-12-18:
Sorry just update the OP.

flothesof, 2022-01-14:
Just a thought, but since this hasn’t gained much traction, I would be in favor of closing this and not doing it.
I’m not sure tofu has an interest to be pixel perfect: from my experience this would create overhead if you try to do this for many usecases. 
A better alternative in my opininion is to have a simulation benchmark that is tested everytime e.g, a high-level outcome of a simulation that has been published in the litterature and that is expected to be simulated correctly by tofu.
E.g. some figure like this one maybe for the inversion algorithms:
[image]
(source: https://iopscience.iop.org/article/10.1088/1742-6596/768/1/012001/pdf)
This sort of thing should be tested not with images but by comparing values as numpy arrays.

Didou09, 2022-01-14:
Agreed !
Thanks, closing this issue then, and maybe the current work in #603 will give us an opportunity to implement this using material that might be published in some time from now.

closed 2022-01-14 by Didou09