ToFuProject/tofu #454: Error for python 3.8 on windows

lasofivec, 2020-12-18:
I think I'm almost all set with the testing part of github actions, except there is one build out of 9 that is failing.
Which is using python 3.8 on windows.

I wonder if this is an issue of the worklfow on GA or if this is due to a bug in `pyinstaller` which apparently is known.

To see the job : https://github.com/lasofivec/tofu/runs/1579478443?check_suite_focus=true

The matrix : [latest versions of mac, windows, linux] x [python 3.6 3.7 3.8]

The log :

=

> ================================= FAILURES ===================================
> ________________________ Test03_Rays.test14_plot_touch ________________________
> 
> self = <tofu.tests.tests01_geom.test_03_core.Test03_Rays object at 0x000001A099CC9F40>
> 
>     def test14_plot_touch(self):
> >       connect = (hasattr(plt.get_current_fig_manager(),'toolbar')
>                    and getattr(plt.get_current_fig_manager(),'toolbar')
>                    is not None)
> 
> tofu\tests\tests01_geom\test_03_core.py:1065: 
> _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
> c:\hostedtoolcache\windows\python\3.8.6\x64\lib\site-packages\matplotlib\pyplot.py:781: in get_current_fig_manager
>     return gcf().canvas.manager
> c:\hostedtoolcache\windows\python\3.8.6\x64\lib\site-packages\matplotlib\pyplot.py:747: in gcf
>     return figure()
> c:\hostedtoolcache\windows\python\3.8.6\x64\lib\site-packages\matplotlib\pyplot.py:687: in figure
>     figManager = new_figure_manager(num, figsize=figsize,
> c:\hostedtoolcache\windows\python\3.8.6\x64\lib\site-packages\matplotlib\pyplot.py:315: in new_figure_manager
>     return _backend_mod.new_figure_manager(*args, **kwargs)
> c:\hostedtoolcache\windows\python\3.8.6\x64\lib\site-packages\matplotlib\backend_bases.py:3494: in new_figure_manager
>     return cls.new_figure_manager_given_figure(num, fig)
> c:\hostedtoolcache\windows\python\3.8.6\x64\lib\site-packages\matplotlib\backends\_backend_tk.py:885: in new_figure_manager_given_figure
>     window = tk.Tk(className="matplotlib")
> _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
> 
> self = <tkinter.Tk object .>, screenName = None, baseName = '__main__'
> className = 'matplotlib', useTk = 1, sync = 0, use = None
> 
>     def __init__(self, screenName=None, baseName=None, className='Tk',
>                  useTk=1, sync=0, use=None):
>         """Return a new Toplevel widget on screen SCREENNAME. A new Tcl interpreter will
>         be created. BASENAME will be used for the identification of the profile file (see
>         readprofile).
>         It is constructed from sys.argv[0] without extensions if None is given. CLASSNAME
>         is the name of the widget class."""
>         self.master = None
>         self.children = {}
>         self._tkloaded = 0
>         # to avoid recursions in the getattr code in case of failure, we
>         # ensure that self.tk is always _something_.
>         self.tk = None
>         if baseName is None:
>             import os
>             baseName = os.path.basename(sys.argv[0])
>             baseName, ext = os.path.splitext(baseName)
>             if ext not in ('.py', '.pyc'):
>                 baseName = baseName + ext
>         interactive = 0
> >       self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
> E       _tkinter.TclError: Can't find a usable init.tcl in the following directories: 
> E           {c:\hostedtoolcache\windows\python\3.8.6\x64\tcl\tcl8.6}
> E       
> E       c:/hostedtoolcache/windows/python/3.8.6/x64/tcl/tcl8.6/init.tcl: couldn't read file "c:/hostedtoolcache/windows/python/3.8.6/x64/tcl/tcl8.6/init.tcl": No error
> E       couldn't read file "c:/hostedtoolcache/windows/python/3.8.6/x64/tcl/tcl8.6/init.tcl": No error
> E           while executing
> E       "source c:/hostedtoolcache/windows/python/3.8.6/x64/tcl/tcl8.6/init.tcl"
[... 87 lines cut ...]
>     warnings.warn("Not implemented yet !, dfit forced to None")
> 
> tofu/tests/tests02_data/test_03_core.py::Test02_DataCam12DSpectral::test15_plot
> tofu/tests/tests02_data/test_03_core.py::Test02_DataCam12DSpectral::test16_compare
>   D:\a\tofu\tofu\tofu\data\_plot.py:1367: RuntimeWarning: All-NaN slice encountered
>     env = [np.nanmin(ldataint[0],axis=0), np.nanmax(ldataint[0],axis=0)]
> 
> -- Docs: https://docs.pytest.org/en/stable/warnings.html
> ============================ slowest 10 durations =============================
> 26.55s call     tofu/tests/tests01_geom/test_03_core.py::Test01_Struct::test16_plot
> 19.90s setup    tofu/tests/tests01_geom/test_03_core.py::Test03_Rays::test00_set_move
> 13.06s call     tofu/tests/tests02_data/test_03_core.py::Test01_DataCam12D::test19_plot_svd
> 12.61s call     tofu/tests/tests01_geom/test_03_core.py::Test03_Rays::test05_strip_nbytes
> 12.42s call     tofu/tests/tests01_geom/test_03_core.py::Test03_Rays::test15_saveload
> 11.55s call     tofu/tests/tests00_root/test_03_plot.py::Test01_plot_shotovervew::test01_plot_shotoverview
> 11.08s call     tofu/tests/tests02_data/test_03_core.py::Test01_DataCam12D::test18_spectrogram
> 10.97s call     tofu/tests/tests01_geom/test_03_core.py::Test01_Struct::test18_saveload
> 9.48s call     tofu/tests/tests02_data/test_03_core.py::Test01_DataCam12D::test17_plot_combine
> 8.38s call     tofu/tests/tests02_data/test_03_core.py::Test01_DataCam12D::test16_compare
> =========================== short test summary info ===========================
> FAILED tofu/tests/tests01_geom/test_03_core.py::Test03_Rays::test14_plot_touch
> =========== 1 failed, 126 passed, 21 warnings in 254.31s (0:04:14) ============
> Error: Process completed with exit code 1.
> 3s
>

Didou09, 2020-12-19:
Hum.... this does not look like it's coming from tofu itself, but rather from an incomplete installation of dependencies or something alike, can you keep an eye on this ? 
However, the equivalent job on Azure (for conda-forge) did not fail, so it may indeed also be due to GA.

flothesof, 2020-12-19:
I think the conda forge build for this configuration also had something
weird (like not running the tests although it should have). This could be
related. Feel free to check if I'm correct.

(note : as it worked as expected for 3.6 and 3.7 I didn't mention it
explicitly)

lasofivec, 2020-12-19:
It worked for the final two builds, so I am not sure if it was just a bug or not

lasofivec, 2020-12-21:
I closed the issue since all the latest builds seem to be working

closed 2020-12-21 by lasofivec