ToFuProject/tofu #463: tofu geom config loading and definition

Didou09, 2021-02-15:
Motivations:
----------------

* The utility tool to load a default tofu config is used quite often and should more readily accessible
* Several config exist in various versions now, and the most complete include PF coils, which are necessary for some uses only

Proposed solutions:
---------------------------
* Implement a shortcut to call the config-loading routine from the root of tofu (so the user does not have to go into tofu.geom.utils)
* Implement a convention: the configuration includes coils only if its name has '-coils' in it
* Move default config definition to a separate dedicated file to make tofu.geom.utils.py lighter and cleanly separate input fro routines

lasofivec, 2021-02-15:
Great idea !
To be clear, we are talking about `tf.geom.utils.create_config` right ?

And for the coils option, what about an optional parameter `with_coils=False` ?

Didou09, 2021-02-16:
Yes, we are talking about tf.geom.utils.create_config.

Good idea too, I don't know which is best honestly.
I have implemented the '-coils' option already (see PR #464 ), I propose we stick to it for a while and see if users complain about it ?