ToFuProject/tofu #471: Wrong input argument definition in _GG.LOS_areVis_PtsFromPts_VesStruct()

Didou09, 2021-02-16:
As described in PR #470 , and as can be seen from the code in tofu/geom/_comp_solidangles.calc_solidangle_particle(), input argument k (=len_v in calc_solidangle_particle()) is necessarily a 2d array because it contains the distance between all points from 2 sets of (N,) and (M,) points respectively, so it is necessarily a (N, M) array.

But the routine in _GG expects a 1d array.

Also, I can't remember why the routine returns an array of int and not bool ?

lasofivec, 2021-02-17:
Answering the last question first : the bool type is kind of complicated to translate from python to numpy to cython (use of int, int8, np.bool,c bool, bint, npy_bool...). They don't use the same and if I try to use bool there a lot of transformations needed which slows down the cython part.

Didou09, 2021-02-17:
> Answering the last question first : the bool type is kind of complicated to translate from python to numpy to cython (use of int, int8, np.bool,c bool, bint, npy_bool...). They don't use the same and if I try to use bool there a lot of transformations needed which slows down the cython part.

OK for me