ToFuProject/tofu #479: Deployment of 1.4.10

Didou09, 2021-03-03:
Deployment of 1.4.10 was only partial (to [Pypi](https://pypi.org/project/tofu/) only, nothing on anaconda, neither on [ToFuProject](https://anaconda.org/tofuproject/tofu) nor on [conda-forge](https://anaconda.org/conda-forge/tofu))

From travis:
-------------
* Branch [1.4.10](https://travis-ci.org/github/ToFuProject/tofu/builds/761193561) did not run all jobs (only 2 linux jobs), but I think a tagged version for deployment should run all tests 
* Deployment was either skip
```
Skipping a deployment with the pypi provider because a custom condition was not met
```
* Or deployment was unsuccesfull:
```
error: could not restore untracked files from stash
The stash entry is kept in case you need it again.
```
* See also issue #478 raised by Florian on the conde-forge issue

From GA:
-----------
* Deployment to test-pypi was unsuccessful
```
HTTPError: 400 Bad Request from https://test.pypi.org/legacy/
Binary wheel 'tofu-1.4.10-cp39-cp39-linux_x86_64.whl' has an unsupported platform tag 'linux_x86_64'.
```
* Deployment to Pypi was [skipped](https://github.com/ToFuProject/tofu/runs/2017222864?check_suite_focus=true)

lasofivec, 2021-03-03:
Travis doesn't taste all the tests on the PR but it does on master, which is exactly the same thing [(see here)](https://travis-ci.org/github/ToFuProject/tofu/builds/761193485). Would you like all the tests to be run on master and on the tagged build ?

>     * Branch [1.4.10](https://travis-ci.org/github/ToFuProject/tofu/builds/761193561) did not run all jobs (only 2 linux jobs), but I think a tagged version for deployment should run all tests

Travis doesn't taste all the tests on the PR but it does on master, which is exactly the same thing [(see here)](https://travis-ci.org/github/ToFuProject/tofu/builds/761193485). Would you like all the tests to be run on master and on the tagged build ?

Didou09, 2021-03-03:
> Travis doesn't taste all the tests on the PR but it does on master, which is exactly the same thing [(see here)](https://travis-ci.org/github/ToFuProject/tofu/builds/761193485). Would you like all the tests to be run on master and on the tagged build ?

Ok, you're right, I agree that it is the same thing and it is not necessary to run all tests on the tagged build. 

But:
----
Are we sure this does not have consequences for the **deployment** ? 
I mean for binary deployment (wheels and conda packages), these binary packages are built and deployed on the test environment. If the tests jobs are not run, then the packages are not built and deployed, right ?
That's fine if Travis doesn't do it (for `conda/ToFuProject`) as long as we are sure conda-forge does it (to `conda/conda-forge`).
But in the current situation none of them works, so we have no binary package deployment.

I guess we can agree on not doing all the tests on the tagged build on Travis provided that the conda-forge deployment does it.
What do you think?
@flothesof , do you know whether the conda-forge deployment did not run because we haven't triggered it yet or because the tests where not done ?

flothesof, 2021-03-03:
Hi @Didou09 
I think you misunderstood what I said about conda-forge: the deployment did run, it is only waiting for approval before getting send to conda-forge. This happens manually, and any of us (Laura, you, me) can do it by approving the PR here: https://github.com/conda-forge/tofu-feedstock/pull/10

The issue I was raising in #478 is: the builds have successfully run (see results here: https://github.com/conda-forge/tofu-feedstock/pull/10/checks), but they ran without running pytest, which entails just modifying the build script. This could mean some problems happened (as always :p).

Since you seem worried about not having 1.4.10 on conda-forge, I'll approve the PR and will submit a modification of the recipe that will run pytest (as the GA CI does IIRC).

Cheers,
Florian

flothesof, 2021-03-03:
I'm switching conda-forge testing to pytest in this PR at the tofu-feedstock: https://github.com/conda-forge/tofu-feedstock/pull/11

A first run gave no errors, so this should be fixed soon.

flothesof, 2021-03-03:
I've just merged this modification, conda-forge now also uses pytest. 1.4.10 is now on conda-forge 🚀 🌟

Didou09, 2021-03-03:
Perfect, thanks a lot @flothesof !

closed 2021-11-15 by Didou09