ToFuProject/tofu #483: tofu compilation vs numpy version - binary incompatibility

Didou09, 2021-03-03:
I think this is a issue we already saw a few weeks ago.

After installation of tofu 1.4.10 on the IRFM intra servers, I tried to import tofu in an ipython console got the error:
```

In [1]: import tofu as tf
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
~/anaconda3/envs/test18/lib/python3.7/site-packages/tofu/geom/_core.py in <module>
     25     import tofu.geom._def as _def
---> 26     import tofu.geom._GG as _GG
     27     import tofu.geom._comp as _comp

~/anaconda3/envs/test18/lib/python3.7/site-packages/tofu/geom/_GG.pyx in init tofu.geom._GG()

ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 88 from C header, got 80 from PyObject

During handling of the above exception, another exception occurred:

ValueError                                Traceback (most recent call last)
<ipython-input-1-33e229938323> in <module>
----> 1 import tofu as tf

~/anaconda3/envs/test18/lib/python3.7/site-packages/tofu/__init__.py in <module>
     60 from tofu.utils import save, load, load_from_imas, calc_from_imas
     61 import tofu._plot as _plot
---> 62 import tofu.geom as geom
     63 from tofu.geom.utils import create_config as load_config
     64 import tofu.data as data

~/anaconda3/envs/test18/lib/python3.7/site-packages/tofu/geom/__init__.py in <module>
     10 """
     11 
---> 12 from tofu.geom._core import PlasmaDomain, Ves, PFC, CoilPF, CoilCS, Config
     13 from tofu.geom._core import Rays, CamLOS1D, CamLOS2D
     14 from tofu.geom._core_optics import *

~/anaconda3/envs/test18/lib/python3.7/site-packages/tofu/geom/_core.py in <module>
     30 except Exception:
     31     from . import _def as _def
---> 32     from . import _GG as _GG
     33     from . import _comp as _comp
     34     from . import _comp_solidangles

~/anaconda3/envs/test18/lib/python3.7/site-packages/tofu/geom/_GG.pyx in init tofu.geom._GG()

ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 88 from C header, got 80 from PyObject
```

After doing some [research](https://ymd_h.gitlab.io/ymd_blog/posts/numpy_1_20_abi_change/), and confirming with the numpy [doc](https://numpy.org/devdocs/release/1.20.0-notes.html#the-pyarray-descrcheck-macro-is-modified), it appears this has to do with the version of numpy installed on the user's environment.

The bottom line is:
* the numpy ABI (binary equivalent of API) has changed since 1.20.0
* If the user has an older version of numpy installed (1.18.1 on IRFM servers), the tofu installation will not work.

I tried it on dedicated test conda environments:
* test18, on which I pre-installed numpy=1.18.1   => pip install tofu leads to the error described above at import time
* test20, on which I pre-installed numpy=1.20.0   => pip install tofu works fine

Question:
-------------

What I do not get is that I get the error on test18, even though tofu is install with pip, from source and compiled locally against the local version of numpy.

I would expect to get the error if I installed tofu from binaries (wheel or conda) because in this case tofu is compiled against a numpy version different from the one installed locally. 
But if it is installed from source and compiled locally against the local numpy version, I don't get why we get the error.

As recommended [here](https://github.com/scikit-learn-contrib/hdbscan/issues/457), I even tried to force local compilation from source using
```
pip install tofu --no-cache-dir --no-binary :all:
```
But the error is still there... I don't get why

Possible workarounds:
------------------------------
* We can force users to use numpy>=1.20.0 by writing it in the dependencies (requirtements.txt, setup.py, meta.yaml)
* But this may be an issue for institutions which have a shared computing server with a team of administrators that want to control the environment (updating numpy just for tofu may cause other issues with other libraries also depending on numpy).
* There is possible more flexible workaround described [here](https://numpy.org/devdocs/reference/c-api/types-and-structures.html#c.PyArrayObject):

> Note
> 
> Further members are considered private and version dependend. If the size of the struct is important for your code, special care must be taken. A possible use-case when this is relevant is subclassing in C. If your code relies on sizeof(PyArrayObject) to be constant, you must add the following check at import time:
> 
> if (sizeof(PyArrayObject) < PyArray_Type.tp_basicsize) {
>     PyErr_SetString(PyExc_ImportError,
>        "Binary incompatibility with NumPy, must recompile/update X.");
>     return NULL;
> }
> 
> To ensure that your code does not have to be compiled for a specific NumPy version, you may add a constant, leaving room for changes in NumPy. A solution guaranteed to be compatible with any future NumPy version requires the use of a runtime calculate offset and allocation size.

lasofivec, 2021-03-03:
We did see this: here #461 
What worked for me was using ` --no-build-isolation` when building. The solution you used did not work for many users.
This specific answer details all the solutions https://github.com/scikit-learn-contrib/hdbscan/issues/457#issuecomment-773671043
And as you can see, there are others that tried your solution (which is written as a possible solution and not a real solution):
https://github.com/scikit-learn-contrib/hdbscan/issues/457#issuecomment-774033215

Could you try this ?

Didou09, 2021-03-03:
Thanks !

It doesn't work if I just use:
```
pip install tofu --no-build-isolation
```

But it does work if I use all 3 options:
```
pip install tofu --no-build-isolation --no-cache-dir --no-binary :all:
```
The way I understand it is:
* `--no-build-isolation`: forces the build to happen in the user's environment (and pip to select the already installed numpy=1.18.1 instead of building against the system's numpy=1.20.0, the problem being that otherwise there is no way for the user to specify which numpy shall be used for compilation)
* `--no-cache-dir`: forces pip to redo the compiling and not use an already existing wheel compiled in the cache memory (because I installed / uninstalled tofu several times in the environment so the cache was not empty)
* `--no-binary`:all:  forces pip to compile and avoid loading wheels

Question:
-------------
So we have a solution, but the user needs to do it himself
* Either we should include this installatiuon procedure in the documentation (for users that run into this error)
* Or we should find a way to automate this

What do you think @lasofivec ?

lasofivec, 2021-03-04:
The thing is that this problem is new and at the same time old as it has happened in previous years.
I don't know how many users install it from source. However for all the machines where tofu is installed it will probably be a problem. I would suggest putting in the doc and sending and email with the instructions ?

Didou09, 2021-03-04:
OK for me, can you put the explanation and workaround in the doc ? I'll send an email to users who face this issue.

Also, maybe later in 2021, we can write numpy >= 1.20.0 in the dependencies, when all users have had time to migrate to newer versions of numpy.

lasofivec, 2021-03-04:
Working on the doc right now.
And yes, the most important is that the clusters get the latest version of numpy

Didou09, 2021-03-04:
Thanks !
Can you use that opportunity to also include the release notes of 1.4.10 in the doc ?

lasofivec, 2021-03-04:
Done in #484

closed 2021-03-09 by Didou09