ToFuProject/tofu #487: ToFu remains stuck when writing the bolometer IDS to disk on ITER HPC

MireilleSchneider, 2021-03-19:
Here is my environment:

```
module purge
module load IPython/6.3.1-intel-2018a-Python-3.6.4
module load ToFu/1.4.10-intel-2018a-Python-3.6.4
module load IMAS
```
Here is the code I execute:

```
import numpy as np
import tofu as tf

multi = tf.imas2tofu.MultiIDSLoader(ids=['edge_sources'],\
             shot=134000,run=30,tokamak='ITER',user='public',ids_base=False,get=False)
multi.add_ids('bolometer', shot=1, run=1, tokamak='ITER', user='schneim', get=True)

conf = tf.geom.utils.create_config("ITER-V0")
cam = multi.to_Cam("bolometer", config=conf, plot=False)

plasma = multi.to_Plasma2D(dsig=['edge_sources'], dextra=False)

sig, units = cam.calc_signal_from_Plasma2D(plasma, quant='edge_sources.2dradiation',Brightness=False,res=1.e-4,plot=False)

sig.save_to_imas(shot=222,run=1,tokamak='ITER',user='schneim',ids='bolometer')
```

.... Then the code remains stuck after created the mdsplus tree file (empty).

Didou09, 2021-06-09:
Debugging ongoing, it seems the user was wrongly interpreted by tofu, and the last updates introduced a new error:

```
In [1]: import tofu as tf
tf./home/ITER/vezined/ToFu_All/tofu/tofu/__init__.py:92: UserWarning: 
The following subpackages are not available:
	- tofu.mag
  => see print(tofu.dsub[<subpackage>]) for details.
  warnings.warn(msg)

In [2]: tf.__file__
Out[2]: '/home/ITER/vezined/ToFu_All/tofu/tofu/__init__.py'

In [3]: conf = tf.load_config('ITER-V0')

In [4]: multi = tf.imas2tofu.MultiIDSLoader(ids='edge_sources', shot=134000, run=30, tokamak='ITER' ,user='public', ids_base=False, get=False)

In [5]: multi.add_ids('bolometer', shot=1, run=1, tokamak='ITER', user='schneim', get=True)
Getting ids   [occ]  tokamak  user     version  shot    run  refshot  refrun
------------  -----  -------  -------  -------  ------  ---  -------  ------
edge_sources  [0]    ITER     public   3        134000  30   -1       -1    
bolometer     [0]    ITER     schneim  3        1       1    -1       -1    

In [6]: cam = multi.to_Cam('bolometer', config=conf, plot=False)

In [7]: plasma = multi.to_Plasma2D(dsig=['edge_sources'], dextra=False)

In [8]: sig, units = cam.calc_signal_from_Plasma2D(plasma, quant='edge_sources.2dradiation', Brightness=False, res=1.e-4, plot=False)
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-13-61a00b856812> in <module>()
----> 1 sig, units = cam.calc_signal_from_Plasma2D(plasma, quant='edge_sources.2dradiation',Brightness=False,res=1.e-4,plot=False)

~/ToFu_All/tofu/tofu/geom/_core.py in calc_signal_from_Plasma2D(self, plasma2d, t, newcalc, quant, ref1d, ref2d, q2dR, q2dPhi, q2dZ, Type, Brightness, interp_t, interp_space, fill_value, res, DL, resMode, method, minimize, num_threads, reflections, coefs, coefs_reflect, ind, returnas, plot, dataname, fs, dmargin, wintit, invert, units, draw, connect)
   7106                 interp_space=interp_space,
   7107                 fill_value=fill_value,
-> 7108                 Type=Type,
   7109             )
   7110 

~/ToFu_All/tofu/tofu/data/_core.py in get_finterp2d(self, quant, ref1d, ref2d, q2dR, q2dPhi, q2dZ, interp_t, interp_space, fill_value, Type)
   4087                                  idq2dPhi=idq2dPhi, idq2dZ=idq2dZ,
   4088                                  interp_t=interp_t, interp_space=interp_space,
-> 4089                                  fill_value=fill_value, ani=ani, Type=Type)
   4090         return func
   4091 

~/ToFu_All/tofu/tofu/data/_core.py in _get_finterp(self, idquant, idref1d, idref2d, idq2dR, idq2dPhi, idq2dZ, interp_t, interp_space, fill_value, ani, Type)
   4024                                                ntall=ntall, mpltri=mpltri,
   4025                                                indtq=indtq, indtr1=indtr1,
-> 4026                                                indtr2=indtr2, trifind=trifind)
   4027 
   4028         return func

TypeError: get_finterp_isotropic() got multiple values for argument 'vquant'
```

Didou09, 2021-06-09:
On branch Issue487:
* The new bug is fixed
* The user-related bug is fixed (original reason for this ticket)

=> one more bug appeared seems to be related to the IDS tree structure

Didou09, 2021-06-09:
Looks like the last bug is not related to tofu but more probably to the IMAS environment, opening a [JIRA tickect](https://jira.iter.org/browse/IMAS-3698?filter=-2) for this.

@MireilleSchneider , once this last IMAS-related bug is cleared (cf. JIRA) you should be able to save synthetic data to IMAS.

What was wrong with your original attempt:
* There was a bug in tofu that did not handle the absolute path to your personal database correctly - **fixed in next release**
* You call to the function was partially incomplete: you should specify a `path_data` argument that is used to specify on which node of the ids you want to save the data (for each channel), in your case `path_data='power.data'`, because in the bolometer ids, each channel has a `power.data` node and that's where you want to store your synthetic data.
* Also, specify `deep=False` to mention that you only want to save the synthetic data, and not the whole problem (i.e.: the tokamak geometry, the bolometer geometry...). By default `deep=True` meaning that tofu attempts a deep save (it tries to save everything). Maybe I'll set it to `deep=False` by default, what do you think?

So, with the next release, and after the fix for the last IMAS-specific issue, the call for saving should be:
```
sig.save_to_imas(shot=222, run=1, tokamak='ITER', user='schneim', ids='bolometer', path_data='power.data', deep=False)
```

closed 2021-11-15 by Didou09