ToFuProject/tofu #488: To enable input database names with underscore

MireilleSchneider, 2021-03-19:
ToFu currently does not work with IMAS input database having underscore, which means that it would not be able to read e.g. the  bolometer IDS directly from the Machine Description database (user='public', database='ITER_MD'). Would it be possible to correct it?

Didou09, 2021-03-23:
Yes, we'll fix it in the next release.
Thanks for reporting

Didou09, 2021-06-09:
@MireilleSchneider I think this issue has already been fixed by another PR, but I would like to check with a realistic test,

would you have a shot/run/user/ to suggest for a test with databes='ITER_MD' ?

Didou09, 2021-06-09:
Done, you can forget about this, I found the following test case that did not pose  problem:
```
In [2]: multi = tf.imas2tofu.MultiIDSLoader(ids='wall', shot=100001, run=1, tokamak='ITER_DISRUPTIONS', user='public', ids_base=False)
Getting ids  [occ]  tokamak           user    version  shot    run  refshot  refrun
-----------  -----  ----------------  ------  -------  ------  ---  -------  ------
wall         [0]    ITER_DISRUPTIONS  public  3        100001  1    -1       -1    
```

I found this on this [webpage](https://confluence.iter.org/display/IMP/Disruption+database), as you can see the data base name has an underscore (`ITER_DISRUPTIONS`) and it works fine.

=> issue considred fixed, will be available in next release, please fell free to comment if you think it is not satisfactory