ToFuProject/tofu #489: Tofu unhappy with specific names for wall.description_2d[0].limiter.unit[0].name

MireilleSchneider, 2021-03-19:
When I execute the following command:

```
multi_edge = tf.imas2tofu.MultiIDSLoader(ids=['edge_sources'],\
             shot=134000,run=30,tokamak='ITER',user='public',ids_base=False,get=False)
multi_edge.add_ids('bolometer', shot=1, run=1, tokamak='ITER', user='schneim', get=True)
multi_edge.add_ids('wall', shot=1180, run=17, tokamak='ITER', user='schneim', get=True)
plasma = multi_edge.to_Plasma2D(dsig=['edge_sources'], dextra=False)
```
... ToFu complains because it is equal to 'First Wall' and it does not like the space. When I replace the name by 'FirstWall' without the space (shot=1180,run=18), then it does not complain anymore and it looks like it reads the wall IDS properly.

Would it be possible to update ToFu such that it accepts names with underscores and spaces?

Didou09, 2021-03-23:
Pull Request #486 solves this by automatically removing spaces and underscores from names inside tofu.

Would this be a satisfactory solution @MireilleSchneider ?
If so, it will be available in the next release (and its already available in the devel branch).

Didou09, 2021-03-23:
However, PR #486 also revealed that the wall ids stores the ITER wall not a closed polygon but as a set of open polygons each describing a different part of the total closed polygon....
This leads tofu to misunderstand the nature of the polygons and requires an extra fix (to be done soon).

MireilleSchneider, 2021-03-23:
If ToFu automatically remove spaces and underscores, I think this is a good fix. As for the inability to read polygons from the wall IDS, I understand that it will be dealt with in another tracker. Thanks!

closed 2021-10-06 by lasofivec