ToFuProject/tofu #521: Creating a geometry from a pdf file through inkscape

lasofivec, 2021-06-29:
I'm creating an issue as it will be better for discussing

Idea:
------
We could use this new svg functionality to show users how they copy copy a published tokamak geometry and easily reproduce it in tofu:
* open inkscape and import an image of a tokmaka poloidal cross-section taken from a published article
* draw the elements you are interested it 
* save as svg
* open ipython, import tofu, load config from svg

What do you think ?
Do you think you could make a tuto video using fig 1 of the article attached (DIIID) and include it in the doc on the same page as the one you just created ? 
[DIIID.pdf](https://github.com/ToFuProject/tofu/files/6733438/DIIID.pdf)

 I don't know yet what the result should be, 
Can you do a try or two trying to get a reasonable representation of the figure ?
It's also an opportunity to figure out what functionality ay be missing to make users' life easier.

It would be nice to get:
* all PF coils
* The double vessel (grey hash on the figure), as 2 vessels
* The innermost vessel (what looks like a unique first wall approximation), as a third vessel

lasofivec, 2021-06-29:
Just a first draft:

https://youtu.be/l6CSlLGDuV0

Didou09, 2021-06-29:
This is already very good 1

The main issue now is how to get the proper scale (important for R in toroidal geometry), since R = 0 does not appear on the axis of the figure.

Any idea ?

Didou09, 2021-06-29:
I propose: the user just draws a line between 2 points of known coordinates, and tells tofu about these coordinates, tofu will find the lin and points, and will deduce the scale and (r0, z0) from it.

I can try an implementation in ~1 hour

lasofivec, 2021-06-29:
what about a bounding box? so a bezier curve with 4 points, with ID = "boundingbox" and when loading the config:
``` 
conf = tf.geom.Config.from_svg("from_pdf.svg", Exp="test", Name="test", boundingbox=[(0.7, -2), (2.8, 2)])
```

Didou09, 2021-06-29:
A bounding box is a good idea, but it requires that at least 4 points are known, whereas the line only requires 2 points and should be sufficient.
Since we do not know in advance what figures will be used and how many points are known, I guess the minimum information is needed, the better.

Can we do less  than 2 points ?
2 points = 4 coordinates
1 points + 1 length = 3 coordinates

Would it make sense to ask for the first point only the length of the line ? that could be sufficient too

lasofivec, 2021-06-29:
do you know if it is possible to draw a single point in Inkscape?

Didou09, 2021-06-29:
Not as far as I know,
So i'm testing the line solution right now, I'll push in ~30 min and you'll tell me if it is practical (let's hope so :-))

closed 2021-10-06 by lasofivec