ToFuProject/tofu #536: Strange Behaviour of solid angle map for particle

Didou09, 2021-07-15:
Motivation:
-------------
As mentionned in PR #535 (there is a starnage behaviour, in some cases, of the solid angle map computation

Proposition:
--------------
* Use the Config.calc_solidangle_particle_integrated() (with the plot), to debug more easily
* Some of it may be due to the fact that the spatial resolution is large compared to the particle, and that if a mesh center is very close to the particle, it will get a very large solid angle (much larger than the other mesh elements, due to the decrease in /dict**2)

... to be investigated

Example:
-----------

```
In [1]: import tofu as tf                                                                                                                                                              
conf = /Home/DV226270/ToFu_All/tofu_git/tofu/tofu/imas2tofu/__init__.py:87: UserWarning: 
You do not seem to be using the latest IMAS version:
'module list' vs 'module av IMAS' suggests:
	- Current version: 3.30.0-4.8.4
	- Latest version : 3.32.1-4.9.1-R2019b
  warnings.warn(msg)

In [2]: conf = tf.load_config('AUG', strict=True)                                                                                                                                      

In [3]: part_radius = np.array([10e-6, 10e-6, 10e-6])                                                                                                                                  

In [4]: part_traj = np.array([[1.8, 0.01, 0.], [1.8, 0.01, -0.1], [1.8, 0.01, -1]]).T                                                                                                                                                           

In [5]: out = conf.calc_solidangle_particle_integrated(part_traj=part_traj, part_radius=part_radius, resolution=0.05, vmax=False)     

```
[image]

What seems wrong in the image above:
in the left plot:

the particle seem to be cenetred around Z=-0.1, which is wrong, it is around Z = 0 (it should be on the red line)
Moreover, the integrated solid angle should be more or less symetric as we go away from the particle, but here there is clear a brutal decrease as we go from the particle towards Z<0 (red cirlce)

lasofivec, 2021-07-20:
it also happens when you try to compute only that test case?

lasofivec, 2021-07-20:
Mmmmm could you tell me what would be the equivalent with my function? So which parameters does it send to `GG.compute_solid_angle_map`. I just tried it and it looks fine but I might not be using the right parameters.

lasofivec, 2021-07-20:
I plot my image using scatter instead of imshow, maybe there is a problem with the plotting?
I used the script below

[image]

```
import tofu as tf
import tofu.geom._GG as GG
import matplotlib.pyplot as plt
import numpy as np

def compute_min_max_r_and_z(ves_poly):
    r_min_max = np.array([np.min(ves_poly[0, :]), np.max(ves_poly[0, :])])
    z_min_max = np.array([np.min(ves_poly[1, :]), np.max(ves_poly[1, :])])
    return r_min_max, z_min_max

conf = tf.load_config('AUG')
part_radius = np.array([10e-6, 10e-6, 10e-6])
part_traj = np.array([[1.8, 0.01, 0.], [1.8, 0.01, -0.1], [1.8, 0.01, -1]],
                     order="F").T

kwdargs = conf.get_kwdargs_LOS_isVis()
ves_poly = kwdargs["ves_poly"]
rstep = zstep = phistep = 0.05
limits_r, limits_z = compute_min_max_r_and_z(ves_poly)
DR = None
DZ = None
DPhi = None
# -- Getting cython APPROX computation --------------------------------
res = GG.compute_solid_angle_map(part_traj, part_radius,
                                 rstep, zstep, phistep,
                                 limits_r, limits_z,
                                 DR=DR, DZ=DZ,
                                 DPhi=DPhi,
                                 block=True,
                                 limit_vpoly=ves_poly,
                                 **kwdargs,
                                 )
pts, sa_map, ind, reso_r_z = res

fig, axs = plt.subplots(1, 3, figsize=(15,15))
for pp in range(3):
    axs[pp].scatter(pts[0, :], pts[1, :],
                    marker="s", edgecolors="None",
                    s=10, c=sa_map[:, pp].flatten(),
                    vmin=sa_map[:, pp].min(),
                    vmax=sa_map[:, pp].max(),
                    )
    axs[pp] = conf.plot(lax=axs[pp], proj='cross', dLeg=None)
plt.show(block=True)
```

Didou09, 2021-07-21:
OK, so maybe it's the plotting then, thanks, I'll double-check !

Didou09, 2021-07-21:
You were right, the problem did not come from your routine.

It came from the intermediate routine that I wrote and that calls your routine, I was passing a non-closed vessel polygon to it, which triggered strange behaviour,
It is fixed now:
```

In [1]: import tofu as tf
/home/didier/Documents/Projects/tofu/tofu/__init__.py:93: UserWarning: 
The following subpackages are not available:
	- tofu.imas2tofu
	- tofu.mag
  => see print(tofu.dsub[<subpackage>]) for details.
  warnings.warn(msg)

In [2]: conf = tf.load_config('AUG')

In [3]: part_traj = np.array([[1.8, 0.01, 0.], [1.8, 0.01, -0.1], [1.8, 0.01, -1]]).T

In [4]: part_radius = np.array([10e-6, 10e-6, 10e-6])

In [5]: ptsRZ, sang, indices, reseff, dax = conf.calc_solidangle_particle_integrated(part_traj=part_traj, part_radius=part_radius, resolution=0.05, vmax=False, approx=False)

```

[image]