ToFuProject/tofu #599: Issues with linux installation

fsciortino, 2021-12-09:
Following the `installing ToFu as a developer` section in https://tofuproject.github.io/tofu/installation.html, I run into some trouble. I tested this on 2 different servers, both with and without conda. Specifically, the error occurs after running `pip install -e .[dev]`. Since I don't have root access on either of the 2 servers that I tested on, in practice I run `python3 -m pip install -e . --user`.

The issue seems to be with the installation trying to write to root-owned directories. The part of the output that seems relevant for debugging, coming after the installation of all dependencies, is the following (here, on the ltxs cluster at IPP-Garching):

```
  Running setup.py develop for tofu
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/afs/ipp-garching.mpg.de/home/s/sciof/tofu/setup.py'"'"'; __file__='"'"'/afs/ipp-garching.mpg.de/home/s/sciof/tofu/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps --user --prefix=
         cwd: /afs/ipp-garching.mpg.de/home/s/sciof/tofu/
    Complete output (43 lines):
    
    Version for setup.py :  1.5.0-61-g2e6e9523
    
    running develop
    WARNING: The user site-packages directory is disabled.
    /usr/lib/python3.6/distutils/extension.py:131: UserWarning: Unknown Extension options: 'language_level'
      warnings.warn(msg)
    /usr/lib/python3.6/distutils/dist.py:261: UserWarning: Unknown distribution option: 'use_scm_version'
      warnings.warn(msg)
    /tmp/pip-build-env-vc1b3n_u/overlay/lib/python3.6/site-packages/setuptools/dist.py:508: UserWarning: The version specified ('1.5.0-61-g2e6e9523') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
      "details." % version
    /tmp/pip-build-env-vc1b3n_u/overlay/lib/python3.6/site-packages/setuptools/command/easy_install.py:159: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
      EasyInstallDeprecationWarning,
    /tmp/pip-build-env-vc1b3n_u/overlay/lib/python3.6/site-packages/pkg_resources/__init__.py:119: PkgResourcesDeprecationWarning: 1.5.0-61-g2e6e9523 is an invalid version and will not be supported in a future release
      PkgResourcesDeprecationWarning,
    /tmp/pip-build-env-vc1b3n_u/overlay/lib/python3.6/site-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
      setuptools.SetuptoolsDeprecationWarning,
    error: can't create or remove files in install directory
    
    The following error occurred while trying to add or remove files in the
    installation directory:
    
        [Errno 13] Permission denied: '/usr/local/lib/python3.6/dist-packages/test-easy-install-12483.write-test'

```

Based on this, my guess is that there's something wrong with some testing capabilities. Unfortunately the `setup.py` is a little involved for me to easily grasp what's going on. Could you please look into this? Thanks!

lasofivec, 2021-12-09:
Hey @fsciortino, I'll look into it. The error occurred when running `pip install -e .[dev]` or `python3 -m pip install -e . --user`?
If it is the first one, are you sure the `pip` being used is the one from your conda install?

lasofivec, 2021-12-09:
Looks like the `--user` is being ignored.

In any case, unless you need to make changes to the `doc` (I don't think this is the case). I would suggest you don't install it in `dev` mode. The `-e .[dev]` will install the dependencies needed for developers, and in most cases it's only `pytest` (and eventually `coverage`) that is needed. So could you try to install it without the `-e` flag?
For example, just using `python setup.py install --user`? just to check if that works

fsciortino, 2021-12-09:
thanks @lasofivec ! The error that I pasted above came from `python3 -m pip install -e .[dev] --user`. 

I've also just tried to create a conda environment specifically for tofu (as advised on the tofu docs), but I get this different error:
```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
omas 0.81.0 requires h5py, which is not installed.
omas 0.81.0 requires netCDF4, which is not installed.
omas 0.81.0 requires pexpect, which is not installed.
ipython 7.16.1 requires backcall, which is not installed.
ipython 7.16.1 requires decorator, which is not installed.
ipython 7.16.1 requires jedi>=0.10, which is not installed.
ipython 7.16.1 requires pexpect; sys_platform != "win32", which is not installed.
ipython 7.16.1 requires pickleshare, which is not installed.
ipython 7.16.1 requires prompt-toolkit!=3.0.0,!=3.0.1,<3.1.0,>=2.0.0, which is not installed.
ipython 7.16.1 requires traitlets>=4.2, which is not installed.
ipyparallel 8.0.0 requires decorator, which is not installed.
ipyparallel 8.0.0 requires entrypoints, which is not installed.
ipyparallel 8.0.0 requires ipykernel>=4.4, which is not installed.
ipyparallel 8.0.0 requires jupyter-client, which is not installed.
ipyparallel 8.0.0 requires psutil, which is not installed.
ipyparallel 8.0.0 requires pyzmq>=18, which is not installed.
ipyparallel 8.0.0 requires tornado>=5.1, which is not installed.
ipyparallel 8.0.0 requires traitlets>=4.3, which is not installed.
astroquery 0.4.3 requires beautifulsoup4>=4.3.2, which is not installed.
astroquery 0.4.3 requires html5lib>=0.999, which is not installed.
astroquery 0.4.3 requires keyring>=4.0, which is not installed.
Successfully installed Jinja2-3.0.3 Pygments-2.10.0 alabaster-0.7.12 babel-2.9.1 certifi-2021.10.8 cycler-0.11.0 docutils-0.17.1 imagesize-1.3.0 packaging-21.3 pluggy-1.0.0 python-dateutil-2.8.2 requests-2.26.0 snowballstemmer-2.2.0 sphinx-4.3.1 sphinxcontrib-applehelp-1.0.2 sphinxcontrib-devhelp-1.0.2 sphinxcontrib-htmlhelp-2.0.0 sphinxcontrib-jsmath-1.0.1 sphinxcontrib-qthelp-1.0.3 sphinxcontrib-serializinghtml-1.1.5 tofu toml-0.10.2
```

fsciortino, 2021-12-09:
@lasofivec the installation in standard (not `dev`) mode did run through, but all the tests (run with `pytest tofu/tests`) seem to be failing within the conda environment :/

lasofivec, 2021-12-09:
When did you get the first error? Was it running `conda create -n tofu3 python scipy numpy cython git ipython`?

fsciortino, 2021-12-09:
The first error was from `python3 -m pip install -e .[dev] --user`. Then, after you suggested not using the developer mode, I used `conda create -n tofu3 python scipy numpy cython git ipython`, following the tofu docs.

lasofivec, 2021-12-09:
I just tried it on my machine and it seems to be working. 
Could you please give me the following info:
- version of conda you are using
- the result of `conda list`
- and the specs of the server you are running so I can try to mount a docker

lasofivec, 2021-12-09:
> The first error was from `python3 -m pip install -e .[dev] --user`. Then, after you suggested not using the developer mode, I used `conda create -n tofu3 python scipy numpy cython git ipython`, following the tofu docs.

Sorry by "first error" I meant:
> `ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.`

But from what I understand it was when you created the environment, so this means it is probably not an issue with `tofu` but with conda. I will look into it.

fsciortino, 2021-12-09:
I'm working on the `lxts` cluster at IPP-Garching -- maybe you have access to it yourself?

I'm using conda 4.10.1 and the `conda list` is pasted below. This is all using the `anaconda/3_2021.05` environment module on `lxts`.

```
(tofu3) sciof@lxts04:~/tofu$ conda list
 packages in environment at /afs/ipp/home/s/sciof/.conda/envs/tofu3:

 Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2021.10.8            ha878542_0    conda-forge
certifi                   2016.9.26                py36_0    conda-forge
charset-normalizer        2.0.9                    pypi_0    pypi
curl                      7.80.0               h2574ce0_0    conda-forge
cython                    0.29.23          py36hc4f0c31_1    conda-forge
expat                     2.4.1                h9c3ff4c_0    conda-forge
gettext                   0.19.8.1          h73d1719_1008    conda-forge
git                       2.34.1          pl5321hc30692c_0    conda-forge
idna                      3.3                      pypi_0    pypi
importlib-metadata        4.8.2                    pypi_0    pypi
krb5                      1.19.2               hcc1bbae_3    conda-forge
ld_impl_linux-64          2.36.1               hea4e1c9_2    conda-forge
libblas                   3.9.0           12_linux64_openblas    conda-forge
libcblas                  3.9.0           12_linux64_openblas    conda-forge
libcurl                   7.80.0               h2574ce0_0    conda-forge
libedit                   3.1.20191231         he28a2e2_2    conda-forge
libev                     4.33                 h516909a_1    conda-forge
libffi                    3.4.2                h7f98852_5    conda-forge
libgcc-ng                 11.2.0              h1d223b6_11    conda-forge
libgfortran-ng            11.2.0              h69a702a_11    conda-forge
libgfortran5              11.2.0              h5c6108e_11    conda-forge
libgomp                   11.2.0              h1d223b6_11    conda-forge
libiconv                  1.16                 h516909a_0    conda-forge
liblapack                 3.9.0           12_linux64_openblas    conda-forge
libnghttp2                1.43.0               h812cca2_1    conda-forge
libnsl                    2.0.0                h7f98852_0    conda-forge
libopenblas               0.3.18          pthreads_h8fe5266_0    conda-forge
libssh2                   1.10.0               ha56f1ee_2    conda-forge
libstdcxx-ng              11.2.0              he4da1e4_11    conda-forge
libzlib                   1.2.11            h36c2ea0_1013    conda-forge
markupsafe                2.0.1                    pypi_0    pypi
ncurses                   6.2                  h58526e2_4    conda-forge
numpy                     1.19.5           py36hfc0c790_2    conda-forge
openssl                   1.1.1l               h7f98852_0    conda-forge
pcre2                     10.37                h032f7d1_0    conda-forge
perl                      5.32.1          1_h7f98852_perl5    conda-forge
pip                       21.3.1             pyhd8ed1ab_0    conda-forge
pyparsing                 3.0.6                    pypi_0    pypi
python                    3.6.15          hb7a2778_0_cpython    conda-forge
python_abi                3.6                     2_cp36m    conda-forge
pytz                      2021.3                   pypi_0    pypi
readline                  8.1                  h46c0cb4_0    conda-forge
scipy                     1.5.3            py36h81d768a_1    conda-forge
setuptools                49.6.0           py36h5fab9bb_3    conda-forge
six                       1.16.0                   pypi_0    pypi
sqlite                    3.37.0               h9cd32fc_0    conda-forge
tk                        8.6.11            h14c3975_1005    omfit
tofu                      1.5.0-61-g2e6e9523           dev_0    <develop>
typing-extensions         4.0.1                    pypi_0    pypi
wheel                     0.37.0             pyhd8ed1ab_1    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
zipp                      3.6.0                    pypi_0    pypi
zlib                      1.2.11            h36c2ea0_1013    conda-forge
```

fsciortino, 2021-12-09:
OK, following our conversation on Zoom, running 
```(tofu3) sciof@lxts04:~/tofu$ conda create -n tofu3 python=3.9 scipy numpy cython^C
(tofu3) sciof@lxts04:~/tofu$ pip install . --user
```
I get
```
Building wheels for collected packages: tofu
  Building wheel for tofu (pyproject.toml) ... done
  Created wheel for tofu: filename=tofu-1.5.0_61_g2e6e9523-cp39-cp39-linux_x86_64.whl size=4109053 sha256=e3057dc2db3732f89874aed0eb03eabc12bac3e7d71219fc2143cfbe25ba37d7
  Stored in directory: /tmp/pip-ephem-wheel-cache-z59jfxgl/wheels/e2/66/48/decf2ae853e6502baf06100d8822c3b7785832c1bc8db3af51
  WARNING: Built wheel for tofu is invalid: Metadata 1.2 mandates PEP 440 version, but '1.5.0-61-g2e6e9523' is not
Failed to build tofu
ERROR: Could not build wheels for tofu, which is required to install pyproject.toml-based projects
```

lasofivec, 2021-12-09:
Ok, my bad `pip install .` is going to expect to find a built wheel which raises the error. To specify a path we need the `-e` flag. Or we could have just done `python setup.py install`.

lasofivec, 2021-12-09:
As for what solved the original problem:
- we erased @fsciortino's conda environment and started with a new one
- `conda create -n tofu3 python=3.9 scipy numpy cython` we had to specify the version of `python`, it was using python 3.10 which seems to have caused some problems.

I don't know what you think @Didou09 but I wouldn't change the doc for this since it might change pretty quickly.

closed 2021-12-09 by lasofivec

fsciortino, 2021-12-09:
Thank you very much for your help @lasofivec !

Didou09, 2021-12-09:
> I don't know what you think @Didou09 but I wouldn't change the doc for this since it might change pretty quickly.

You're rigth @lasofivec, it was a minor bug after all, since we have this issue tracker to be used as reference if the problem happens again, I agree that it might not be worth changing the doc.
Thanks for your very fast and efficient reaction !

flothesof, 2021-12-09:
Even though the python versions change, I think it's worth putting in a reference number like 3.9.
Right now, conda forge only builds up to that version anyway.
From my point of view it's a worthwhile change of the doc (and I don't really see a downside to it).
Furthermore the zen of python says : implicit is better than implicit 😬

flothesof, 2021-12-09:
I wouldn't mind going over the install instructions again and submitting a PR, if you guys agree.