ToFuProject/tofu #608: Installing as a developer

fsciortino, 2022-01-11:
@Didou09 I started looking at your latest tutorial (https://github.com/ToFuProject/tofu/issues/603#issuecomment-1009929042), but I am struggling to get the `devel` branch to work/install. It doesn't seem to be the same problem that I run into some time ago, and that @lasofivec helped me with. The new error is shown below.

To reproduce the error: I followed exactly the instructions in the "Installing tofu as a developer" section of the docs:  https://tofuproject.github.io/tofu/installation.html#installing-tofu-as-a-developer. I created a branch new conda environment, and everything else as written in that docs section. I'm working on the lxts cluster at IPP. 

Sorry for this rather annoying/boring issue...

<details>
  <summary>Click to expand!</summary>

  (tofu3new) sciof@lxts00:\~\tofu$ pip install -e .[dev] --user
  Obtaining file:///afs/ipp-garching.mpg.de/home/s/sciof/tofu
    Installing build dependencies ... done
    Checking if build backend supports build_editable ... done
    Getting requirements to build wheel ... done
    Preparing metadata (pyproject.toml) ... done
  Requirement already satisfied: requests in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from tofu===1.5.0-80-g4f0625a0) (2.26.0)
  Requirement already satisfied: numpy in /afs/ipp-garching.mpg.de/home/s/sciof/.conda/envs/tofu3new/lib/python3.9/site-packages (from tofu===1.5.0-80-g4f0625a0) (1.22.0)
  Requirement already satisfied: svg.path in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from tofu===1.5.0-80-g4f0625a0) (4.1)
  Requirement already satisfied: scipy in /afs/ipp-garching.mpg.de/home/s/sciof/.conda/envs/tofu3new/lib/python3.9/site-packages (from tofu===1.5.0-80-g4f0625a0) (1.7.3)
  Requirement already satisfied: cython>=0.26 in /afs/ipp-garching.mpg.de/home/s/sciof/.conda/envs/tofu3new/lib/python3.9/site-packages (from tofu===1.5.0-80-g4f0625a0) (0.29.26)
  Requirement already satisfied: matplotlib in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from tofu===1.5.0-80-g4f0625a0) (3.5.0)
  Requirement already satisfied: sphinx-gallery in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from tofu===1.5.0-80-g4f0625a0) (0.10.1)
  Requirement already satisfied: pytest in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from tofu===1.5.0-80-g4f0625a0) (6.2.5)
  Requirement already satisfied: check-manifest in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from tofu===1.5.0-80-g4f0625a0) (0.47)
  Requirement already satisfied: coverage in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from tofu===1.5.0-80-g4f0625a0) (6.2)
  Requirement already satisfied: sphinx in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from tofu===1.5.0-80-g4f0625a0) (4.3.1)
  Requirement already satisfied: sphinx-bootstrap-theme in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from tofu===1.5.0-80-g4f0625a0) (0.8.0)
  Requirement already satisfied: toml in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from check-manifest->tofu===1.5.0-80-g4f0625a0) (0.10.2)
  Requirement already satisfied: build>=0.1 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from check-manifest->tofu===1.5.0-80-g4f0625a0) (0.7.0)
  Requirement already satisfied: setuptools in /afs/ipp-garching.mpg.de/home/s/sciof/.conda/envs/tofu3new/lib/python3.9/site-packages (from check-manifest->tofu===1.5.0-80-g4f0625a0) (60.5.0)
  Requirement already satisfied: packaging>=20.0 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from matplotlib->tofu===1.5.0-80-g4f0625a0) (21.3)
  Requirement already satisfied: python-dateutil>=2.7 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from matplotlib->tofu===1.5.0-80-g4f0625a0) (2.8.2)
  Requirement already satisfied: fonttools>=4.22.0 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from matplotlib->tofu===1.5.0-80-g4f0625a0) (4.28.3)
  Requirement already satisfied: cycler>=0.10 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from matplotlib->tofu===1.5.0-80-g4f0625a0) (0.11.0)
  Requirement already satisfied: kiwisolver>=1.0.1 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from matplotlib->tofu===1.5.0-80-g4f0625a0) (1.3.2)
  Requirement already satisfied: pillow>=6.2.0 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from matplotlib->tofu===1.5.0-80-g4f0625a0) (8.4.0)
  Requirement already satisfied: setuptools-scm>=4 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from matplotlib->tofu===1.5.0-80-g4f0625a0) (6.3.2)
  Requirement already satisfied: pyparsing>=2.2.1 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from matplotlib->tofu===1.5.0-80-g4f0625a0) (3.0.6)
  Requirement already satisfied: pluggy<2.0,>=0.12 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from pytest->tofu===1.5.0-80-g4f0625a0) (1.0.0)
  Requirement already satisfied: iniconfig in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from pytest->tofu===1.5.0-80-g4f0625a0) (1.1.1)
  Requirement already satisfied: attrs>=19.2.0 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from pytest->tofu===1.5.0-80-g4f0625a0) (21.2.0)
  Requirement already satisfied: py>=1.8.2 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from pytest->tofu===1.5.0-80-g4f0625a0) (1.11.0)
  Requirement already satisfied: certifi>=2017.4.17 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from requests->tofu===1.5.0-80-g4f0625a0) (2021.10.8)
  Requirement already satisfied: charset-normalizer~=2.0.0 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from requests->tofu===1.5.0-80-g4f0625a0) (2.0.9)
  Requirement already satisfied: idna<4,>=2.5 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from requests->tofu===1.5.0-80-g4f0625a0) (3.3)
  Requirement already satisfied: urllib3<1.27,>=1.21.1 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from requests->tofu===1.5.0-80-g4f0625a0) (1.26.7)
  Requirement already satisfied: Pygments>=2.0 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from sphinx->tofu===1.5.0-80-g4f0625a0) (2.10.0)
  Requirement already satisfied: sphinxcontrib-qthelp in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from sphinx->tofu===1.5.0-80-g4f0625a0) (1.0.3)
  Requirement already satisfied: alabaster<0.8,>=0.7 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from sphinx->tofu===1.5.0-80-g4f0625a0) (0.7.12)
  Requirement already satisfied: sphinxcontrib-serializinghtml>=1.1.5 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from sphinx->tofu===1.5.0-80-g4f0625a0) (1.1.5)
  Requirement already satisfied: imagesize in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from sphinx->tofu===1.5.0-80-g4f0625a0) (1.3.0)
  Requirement already satisfied: docutils<0.18,>=0.14 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from sphinx->tofu===1.5.0-80-g4f0625a0) (0.17.1)
  Requirement already satisfied: sphinxcontrib-devhelp in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from sphinx->tofu===1.5.0-80-g4f0625a0) (1.0.2)
  Requirement already satisfied: sphinxcontrib-htmlhelp>=2.0.0 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from sphinx->tofu===1.5.0-80-g4f0625a0) (2.0.0)
  Requirement already satisfied: sphinxcontrib-applehelp in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from sphinx->tofu===1.5.0-80-g4f0625a0) (1.0.2)
  Requirement already satisfied: babel>=1.3 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from sphinx->tofu===1.5.0-80-g4f0625a0) (2.9.1)
  Requirement already satisfied: Jinja2>=2.3 in /afs/ipp-garching.mpg.de/home/s/sciof/.local/lib/python3.9/site-packages (from sphinx->tofu===1.5.0-80-g4f0625a0) (3.0.3)
[... cut ...]
  ERROR tofu/tests/tests04_spectro/test_01_fit12d.py
  ERROR tofu/tests/tests04_spectro/test_02_analysistools.py
  ERROR tofu/tests/tests05_nist/test_03_core.py
  ERROR tofu/tests/tests06_mesh/test_01_checks.py
  ERROR tofu/tests/tests07_inversions/test_01_isotropic.py
  ERROR tofu/tests/tests09_tutorials/test_01_runall.py
  !!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 15 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!
  ====================================== 15 errors in 8.25s =======================================

</details>

lasofivec, 2022-01-12:
Hey @fsciortino, looks like the installation went well so I have the feeling that it is not looking in the right place for the installed version of tofu.
Would you mind doing `cd tofu; pytest tests` .
I am also curious about what branch did we tested together last time? The default branch you get when cloning the rep is `devel` so I would have imagined we installed it and tested it last time.

flothesof, 2022-01-12:
I just checked the developer instructions and was able to install correctly on windows.

@fsciortino, I noticed you used the --user flag for pip in your log.

```
pip install -e .[dev] --user
```
This might cause problems (but I’m no expert). @lasofivec did you install like this last time too?

Cheers,
Florian

lasofivec, 2022-01-12:
> This might cause problems (but I’m no expert). @lasofivec did you install like this last time too?

I don't think we did but can say for sure. I would say that python is looking for the wrong location for tofu and is using firstly the local directory tofu instead of the installation therefore it doesn't find the compiled library GG. But that's just my hunch

flothesof, 2022-01-12:
Strange.
Maybe using "python setup.py install" would work? Worth trying...

fsciortino, 2022-01-12:
thank you both for this help/feedback. I was on a quick zoom call with @Didou09 just a few mins ago and we found that there was a nasty interaction between a version of tofu in my `.local` path (probably left from previous tests with pip) and the `devel` installation that I was doing yesterday/today starting from the cloned directory. 

In other words, I think that I messed up, and it wasn't tofu's fault :/

With this, I learnt that pip and setuptools are pretty bad at understanding multiple sources of installations. In my case, python was trying to fetch some libraries from the pip installation and others from the cloned directory. Just a mess....

Anyway, the solution was to delete everything related to tofu from my `.local` directory and then the cloned `devel` directory worked fine! I think that I'm all set. Sorry for taking your time and many thanks for coming to the help!

closed 2022-01-12 by fsciortino

Didou09, 2022-01-12:
Thanks very much @lasofivec and @flothesof for your continuous support !  :-)