ToFuProject/tofu #662: Make Polygon3 available on conda-forge conda channel

Didou09, 2022-07-28:
The following library is a new dependency for `tofu`: `Polygon3`

It is currently available on `Pypi`, but not on the anaconda repo (at least not in `conda-forge`)
https://pypi.org/project/Polygon3/ 

=> Create a `recipe` and `feedstock` for this library to be on `conda-forge`

Didou09, 2022-10-11:
Hi @flothesof , any chanc you can have a look into this ? ;-)

flothesof, 2022-10-15:
Hi @Didou09,
I would be interested to contribute to this. However, maybe this should also be discussed with the package owner first?
Appartently it is on github: https://github.com/jraedler/Polygon3
Cheers,
Florian

Didou09, 2022-11-16:
Hi @flothesof , I feel like the owner is not active anymore on Github, his last post dates 2 years back, and he hasn't responded to any of the issues opened in his repo.

I feel like we should move on to creating the conda-feedstock to put it on conda-forge, I'm not sure I can see reasons why he would not want it.
What do you think ?

A quick search on Anaconda reveals that several users have forked it and created their own conda version of it:
https://anaconda.org/search?q=Polygon

In this respect, putting Polygon3 on conda-forge would make it easily accessible to all conda users.

flothesof, 2022-11-18:
Hi @Didou09 
I just checked the latest tofu builds on conda-forge. They are failing due to Polygon missing. Maybe it is the right time to solve this issue then :)
I’ll try to have a look and report back.
Cheers,
Florian

flothesof, 2022-11-18:
I’ve been able to progress on this. 
A pull request for merging into conda-forge is on its way here: https://github.com/conda-forge/staged-recipes/pull/21195
I used grayskull as recommended in the docs to generate a recipe and have proposed it to conda-forge.
An encouraging result is that the result ran on the CI without incidents.

flothesof, 2022-11-20:
Right now, I'm waiting for a review of the PR. Given that conda-forge is community operated, I guess this can take a while.

flothesof, 2022-12-08:
Quick update here: I had a first round of reviews and am awaiting the second round. Hopefully soon.

obrejank, 2022-12-09:
#715 removes most of the calls to Polygon3, and what's left will be easy to remove. Sorry for pulling the rug out from you @flothesof.