ToFuProject/tofu #697: plot_geometry_matrix with indbf provided

tbarbui, 2022-10-26:
The function plot_geometry_matrix when called with parameter indbf generates an IndexError (index out of bounds) if indbf>max(channel).
My impression is that indbf runs over the channel indexes instead that over the basic function indexes.

Didou09, 2022-10-26:
Thanks Tullio, will have a look soon,

Could you provide a minimum working example so I can reproduce the error ?

tbarbui, 2022-10-26:
I was reproducing your script for the inversions (see attachment).
When I execute the function plot_geometry_matrix adding the index of the basic function to be plotted, it generates an IndexError if indbf>max(channel).

dax = plasma.plot_geometry_matrix(key='matrix0', res=0.01, indbf=500)

Thanks for you help!

[tofu for inversions.zip](https://github.com/ToFuProject/tofu/files/9869966/tofu.for.inversions.zip)

Didou09, 2022-11-07:
Which version of tofu are you using ?
It works for me...

[image]

Successfully tested with tofu `1.6.2` and `1.6.3`, as well as on the current `devel` branch.

Are you sure you did not redefine 'matrix0' as something else ?
The error msg suggests that in you case the mesh does not have enough bsplines (you're asking for index 500, so if the mesh has than 500 bsplines you will get the error msg you mention).

You can check the size if the geometry matrices by just typing:
```
In [23]: plasma.show('matrix')
matrix   bsplines  cam   data     crop   shape        
-------  --------  ----  -------  -----  -------------
matrix0  m0-bs2    cam0  matrix0  True   (100, 1991)  
matrix1  m1-bs2    cam0  matrix1  False  (20, 100, 9) 
matrix2  m2-bs2    cam0  matrix2  False  (20, 100, 12)
```
Which shows that geometry matrix `matrix0` is a (100, 1991) 2d array
It has 100 lines (i.e.: channels) and 1991 bsplines.
So here asking for `indbf=500` does work.

Try to double-check that and tell me what you discover

tbarbui, 2022-11-07:
Hi Didier,

I'm using tofu 1.6.3.
You're right that it works fine for the example I gave you. Sorry, last time I tried I must have done something silly cause it was giving me that error...
However if I try it with the script here below (taken from Issue #603 Get started on AUG) the error appears.

`conf = tf.load_config('AUG-V0')
cam0 = tf.geom.utils.create_CamLOS1D(pinhole=[1.48, 0, -1.15], orientation=[0,0,0], sensor_nb=10, sensor_size=0.01, focal=0.02, config=conf, Name='test', Diag='test', Exp='AUG')
cam1 = tf.geom.utils.create_CamLOS1D(pinhole=[1.47, 0, -1.10], orientation=[0,0,0], sensor_nb=10, sensor_size=0.01, focal=0.02, config=conf, Name='test', Diag='test', Exp='AUG')
cam2 = tf.geom.utils.create_CamLOS1D(pinhole=[1.46, 0, -1.10], orientation=[0,0,0], sensor_nb=10, sensor_size=0.01, focal=0.02, config=conf, Name='test', Diag='test', Exp='AUG')
cam = cam0 + cam1 + cam2
cam.plot()

mesh = tf.data.Plasma2D()
mesh.add_mesh(domain=[[1.45, 1.65], [-1.22, -1.02]], crop_poly=conf, res=0.008, deg=0, key='mesh1')
mesh.add_geometry_matrix(cam=cam, res=0.001, key='mesh1-bs0')

dax = mesh.plot_geometry_matrix(key='matrix0', indbf=50, indchan=15, cam=cam)`

If I print out the matrix:

`In [64]: mesh.show('matrix')
matrix   bsplines   cam             data     crop  shape    
-------  ---------  --------------  -------  ----  ---------
matrix0  mesh1-bs0  test+test+test  matrix0  True  (30, 365)`

Didou09, 2022-11-07:
OK, I've found the [bug](https://github.com/ToFuProject/datastock/pull/59#issue-1438728233).
It was not in tofu itself but in one of its dependencies, [datastock](https://github.com/ToFuProject/datastock), that I also develop.

The bugfix will be available with the next datastock release, which should also include units management (using astropy).

In the meantime can you live with this bug ?
It's minor and it just means you cannot visualize bsplines of index larger than the number of channels (and vice-versa).
Since this is a minor feature, just set indbf=0 and it will be fine.

You'll receive a notification when datastock is updated (and tofu will be updated accordingly)

Thanks for the feedback and bug-hunting ! :-)

tbarbui, 2022-11-08:
Thanks a lot Didier for finding the bug.
Of course I can live with it, it's a minor bug.

Cheers.

closed 2022-11-23 by dvezinet