ToFuProject/tofu #729: Etendue computation

tbarbui, 2023-03-02:
Hi Didier,

I was comparing the etendue calculated by tofu using the function add_camera_pinhole with my own calculation using this simple formula:

`etendue = Apin*Apxl/d^2*cosd(theta).^4; [m^2*str]`

where:
`
alpha = 19.5; %vertical extension of the viewing solid angle
theta = linspace(alpha/2, -alpha/2, 487);
Apin = 4.e-6; %m^2
Apxl = 0.172^2*1e-6;
d = 0.24744; %distance pinhole-detector
etendue = Apin*Apxl/d^2*cosd(theta).^4; %m^2*str`

The two calculations vary by a factor of three, with yours being the largest.
I double checked the units (both etendues are in m^2*str) and the camera parameters.

I was wondering if your computation is somehow more complex than my simple formula and therefore gives different results.

For the sake of completeness, here are the camera parameters I use in tofu:

```
coll.add_camera_pinhole(
        # naming
        key='c10',             # of the camera
        key_pinhole='p10',
        key_diag='mehxr',      # created if doesn't exist, if it already exists, camera will be added to it
        # cam type
        cam_type='2d',
        # pinhole center       # the pinhole is taken as the center of the aperture
        x=5.1983,
        y=-0.166,
        z=0.205,
        # orientation
        theta=3.25159,         # in a poloidal cross-section, theta = 0 means normal vector = eR
        dphi=-0.0319064,       # angular shift with respect to poloidal cross-section (dphi = 0 => stays in poloidal cross-section)
        tilt=0.,               # angular tilt around the camera's own axis
        # focal distance
        focal=0.2474,          # distance between aperture and pixel plane
        # pixels
        pix_nb=[195, 487],     # Nb. of pixels in [hor., vert.] directions
        pix_size=172e-6,       # square pixels are assumed
        pix_spacing=0.,        # additional spacing between pixels
        # pinhole / slit
        pinhole_size=[4.e-3, 1.e-3], # 'pinhole_radius' for a circular pinhole, 'pinhole_size' for a rectangular aperture
        # los-computation
        config=conf,           # the tokamak geometry is used to ray-tracing (computing LOS)
        reflections_nb=0,      # optionnally add n specular reflections
    )

```

Thank you,
Tullio

Didou09, 2023-03-03:
Hi @tbarbui ,

Thanks for that feedback.

This is simply due to a confusing definition of aperture_size.
When I coded it, I assumed the user would use it interchangeably with the pinhole radius (which is the half of the characteristic length of a circular aperture).
So the argument aperture_size is used in the code like if it was the half-size (like the radius).

You can find the way tofu computes the outline of the aperture from the size in file tofu/data/_class7_compute.py, lines 328-329:
[image]

These are the outline in direction x0 and x1 respectively.
As you can see, the size given in the argument is effectively the half-size.

I agree that this is confusing, I'll fix it so it is used as full-size in future versions.
Sorry I should have made it clearer.
That's were your factor 4 was coming from (2^2).

You can check by viewing the area of the aperture that tofu computes:
```
(Pdb) coll.dobj['aperture']['p10']['dgeom']['area']
1.6e-05
```

And the outline:
```
(Pdb) coll.dobj['aperture']['p10']['dgeom']['outline']
('p10-x0', 'p10-x1')
(Pdb) coll.ddata['p10-x0']['data']
array([-0.004,  0.004,  0.004, -0.004])
(Pdb) coll.ddata['p10-x1']['data']
array([-0.001, -0.001,  0.001,  0.001])
```

tbarbui, 2023-03-03:
Thanks Didier for the explanation. I will modify my script accordingly.

Cheers,
Tullio