ToFuProject/tofu #731: Pip installation failing with ModuleNotFound error (numpy)

jlball, 2023-03-24:
Attempting to install tofu by running `pip install .` and by following the developer installation instructions in the documentation yields the following error:
[image]

And here is the specific output for attempting installation from a conda environment:
[image]

Didou09, 2023-03-31:
Hi @jlball , thanks for reporting this.

Can you try a pip install from the official Pypi repo:

`pip install tofu`

Your screenshot indicate that you attempted a pip install from a pip package that you likely created yourself locally (from the devel branch ?), which is a process I cannot control.
Please try first with the official Pypi repo and re-open this issue if the problem persists (please also include screenshots in that case).

Didou09, 2023-03-31:
General guidelines:
--------------------------

* Always attempt first to install an official release (using `pip` from [Pypi](https://pypi.org/project/tofu/) or using `conda` from [anaconda.org](https://anaconda.org/conda-forge/tofu) if the version of tofu you are looking for is available there).

* If not working, fall back to installing from source (clone) from the `master` branch
     -  the `master` branch contains official releases only. It should be identical code to the latest official release found on `Pypi`

* Only use the `devel` branch if you know what you are doing. While it is relatively well-tested, it is not an official release but rather work in progress and is less tested / can be unstable.

However, I'd like to thank you because your attempt at installing from `devel` and your reporting this issue helped me identify a bug that was introduced since the last official version (numpy had been removed from pyproject.toml, which explains your error message).
The cause for this issue has thus been addressed in PR #

Didou09, 2023-03-31:
General guidelines:
--------------------------

* Always attempt first to install an official release (using `pip` from [Pypi](https://pypi.org/project/tofu/) or using `conda` from [anaconda.org](https://anaconda.org/conda-forge/tofu) if the version of tofu you are looking for is available there).

* If not working, fall back to installing from source (clone) from the `master` branch
     -  the `master` branch contains official releases only. It should be identical code to the latest official release found on `Pypi`

* Only use the `devel` branch if you know what you are doing. While it is relatively well-tested, it is not an official release but rather work in progress and is less tested / can be unstable.

However, I'd like to thank you because your attempt at installing from `devel` and your reporting this issue helped me identify a bug that was introduced since the last official version (numpy had been removed from pyproject.toml, which explains your error message).
The cause for this issue has thus been addressed in PR #736