ToFuProject/tofu #748: Minor bugs observed in compute_diagnostic_signal()

cjperks7, 2023-04-14:
I found a few things that seem like small bugs in compute_diagnostic_signal() routine

1) In `_class8_compute_signal._compute_los()` if "spectro=True", key_cam wasn't being passed into `get_diagnostic_lamb()`. See the attached screenshot. Note that I used key_cam = key_cam[0] because `compute_diagnostic_signal()` accepts key_cam (and even turns into if initially a string) as a list while `get_diagnostic_lamb()` doesn't. I do all the cameras at the same time and conveniently the wavelength basis is the same for each, but this isn't the most general fix
[image]

2) I've been running this function with `compute_diagnostic_signal(brightness=True)` to compare the results of each step with Tofu to the old workflow. It seems like the units for the brightness is wrong though. I input a spectral-resolved emissivity field with units ph/s/m3/sr/m whereas the units for the brightness as shown with `diag.show('data')` is in ph/s/m3/sr. Upon inspection of the code, it lost a unit of meter because of spectral binning. But wouldn't it lose another unit of meter from the LOS integration, so the ultimate unit is ph/s/m2/sr?

cjperks7, 2023-04-14:
3) Going to just document it here while I'm at it. I'm observing the the brightness from Tofu is about an order of magnitude higher than the brightness values I got from the old workflow. I'll do a more quantitative analysis soon

Didou09, 2023-04-14:
I think it may come from the way the etendue is calculated, to be checked vs your analysis and vs the reverse ray-tracing I'm currently finishing

cjperks7, 2023-04-14:
But when brightness=True, it looks like multiplying by the etendue is skipped so I don't have to consider it right now

cjperks7, 2023-04-19:
I resolved my issue in (2) myself. I just hadn't defined units for my (R,Z) mesh so that my fault

Didou09, 2023-04-20:
> I resolved my issue in (2) myself. I just hadn't defined units for my (R,Z) mesh so that's my fault

Yes, Brightness = True essentially just does LOS-integration, and the units are picked up from the (R, Z) units

Didou09, 2023-04-20:
> But when brightness=True, it looks like multiplying by the etendue is skipped so I don't have to consider it right now

Yes, etendue is not used for Brightness = True

Didou09, 2023-04-20:
> 

OK, so am I understanding correctly that you have just validated tofu's LOS-integration + spectral binning by comparison with your homebrew code using that fundamental theorem of calculus ?

Am I also understanding correctly that the difference observed when you opened the issue was essentially coming from the fact that your workflow was not doing the energy binning correctly right ?

cjperks7, 2023-04-20:
> > 
> 
> OK, so am I understanding correctly that you have just validated tofu's LOS-integration + spectral binning by comparison with your homebrew code using that fundamental theorem of calculus ?
> 

Yes, this is validating the LOS-integration & spectral binning steps for the same input data and simulation settings. Recall that we went compared my numbers ~December just using back-of-the-envolope methods and it agreed, so I'm essentially doing the same thing here.

> Am I also understanding correctly that the difference observed when you opened the issue was essentially coming from the fact that your workflow was not doing the energy binning correctly right ?

I think my workflow is doing the energy binning correctly, but I don't understand why the bins used in ToFu are 4.3x larger than I'd expect just looking at the spacing in the wavelength mesh. We use basically the exact same wavelength mesh so the bottom plot is showing that the LOS-integration step seems to be the same for both workflows, but the brightness profiles differ by 4.3x because ToFu is getting it's energy bins too wide for some reason

Didou09, 2023-04-20:
OK thanks, I'll investigate why the bins are larger in tofu