ToFuProject/tofu #755: Planar crystal: provide extenthalf vs outline ?

dvezinet, 2023-05-04:


cjperks7, 2023-05-04:
I've noticed that when building a planar crystal spectrometer that when you use `diag.add_crystal()`, it requires the crsytal['dgeom'] to not have an 'extenthalf' entry:
[image]

But, if you don't define an 'extenthalf' (so just 'outline_x0', 'outline_x1') then later in `diag.add_diagnostic()` you get an error basically requesting it:
[image]

I've been able to work around this by manually adding 'extenthalf' after using `diag.add_crystal()`

cjperks7, 2023-05-04:
Minimum working code:
[make_tofu_spec.txt](https://github.com/ToFuProject/tofu/files/11400287/make_tofu_spec.txt)

closed 2023-09-18 by Didou09