ToFuProject/tofu #830: Trouble Installing Tofu version 1.7.0 with pip

shonmackie, 2023-09-15:
I tried installing Tofu v1.7.0 via pip through vscode terminal on my windows laptop. I am using python v3.10.11. I get the error message in the attached .txt file:
[Tofu_error.txt](https://github.com/ToFuProject/tofu/files/12623605/Tofu_error.txt)

I'm using numpy v1.22.4, and cython v0.29.33

dvezinet, 2023-09-15:
OK, thanks @shonmackie , I know these errors, they are recent and are related to that new version of Cython (one of the dependency of tofu) that mentionned earlier in my email.

I've already fixed that and the bugfix will be available in the next release (1.7.1), that I hope to publish early next week (maybe this week-end if I'm motivated enough).

I you're in such a urry that you can;t wait a few days, you can try to downgrade your version of cython to 0.28 (or 0.27) and see if it fixes it (it should), but I'd simply recommend to wait a bit (as downgrading a library can also trigger other unforeseen dependency issues)

surajgupta4, 2023-09-16:
I am also having same error in installing tofu.

closed 2023-09-18 by Didou09