ToFuProject/tofu #850: Single-channel 1d camera does not seem to work

dvezinet, 2023-09-25:
The following MWE creates a single one-channel camera that does not seem to compute:

```
import tofu as tf
import numpy as np
import matplotlib.pyplot as plt

inch    = 2.54e-2; # m
foot    = 12*inch; # m
#load simple SPARC geometry
config_SPARC = tf.load_config('SPARC-V0')
#load emissivity profiles from TRANSP simulation
r_grid = np.linspace(1, 2.4, 50)
z_grid = np.linspace(-1, 1, 50)
emiss2d = np.exp(-(r_grid[:, None] - 2.0)**2/0.3**2 - (z_grid[None, :] - 0)**2/0.4**2)

#define some parameters
col_r = 0.015 #m
det_r = 0.015 #m
d_cf = 0.30 #foil 10cm back from collimator
pos_det = np.r_[14.14428527845,0.40211187714, 0.1875] #detector center position
nin_det = np.r_[0, -0.34921, -0.017455] #detector LOS unit vector
nin_det = nin_det/np.linalg.norm(nin_det)
e0 = np.r_[-nin_det[1], nin_det[0], 0]
e0 = e0/np.linalg.norm(e0)
e1 = np.cross(nin_det, e0)
param = np.linspace(0,2*np.pi,100) #parameter for defining geometries

#Tofu uses collection objects for just about everything
U238=tf.data.Collection()
U238.add_mesh_2d_rect(key='m0', domain=None, knots0=r_grid, knots1=z_grid, crop_poly=config_SPARC, units =['m','m'])
U238.add_bsplines(key='m0',deg=1)
U238.add_data(key='emiss2d', data=emiss2d*(5/3.95)/(4*np.pi), ref='m0',units='ph/(m3.sr)')#normalize emissivity to correct units0

dfoil = {
    'cents_x': pos_det[0],
    'cents_y': pos_det[1],
    'cents_z': pos_det[2],
    'nin': nin_det,
    'e0': e0,
    'e1': e1,
    'outline_x0': det_r*np.cos(param),
    'outline_x1': det_r*np.sin(param)
}

U238.add_camera_1d(key='det', dgeom=dfoil)

dap1 = {
    'outline_x0': col_r*np.cos(param),
    'outline_x1': col_r*np.sin(param),
    'cent': pos_det+d_cf*nin_det,
    'nin': nin_det,
    'e0': e0,
    'e1': e1
}

U238.add_aperture(key='ap1', **dict(dap1))

sh_r = 0.342 #shield hole dimension

dap2 = {
    'outline_x0': col_r*np.cos(param),
    'outline_x1': col_r*np.sin(param),
    'cent': np.r_[3.894, 0, 0],
    'nin': [-1,0,0],
    'e0': [0,-1,0],
    'e1': np.cross([-1,0,0], [0,-1,0])
}

U238.add_aperture(key='ap2', **dict(dap2))

optics =['det', 'ap1', 'ap2']
doptics = {'det': ['ap1', 'ap2']}

U238.add_diagnostic(
    key='U238',
    doptics=optics,
    compute=True,
    config=config_SPARC
)

dax = U238.plot_diagnostic(key='U238', elements='o', plot_config=config_SPARC)
plt.savefig('GeometryTest.PNG')

U238.compute_diagnostic_signal(
    key='synth_flux',
    key_diag='U238',
    key_integrand='emiss2d',
    brightness=False,
    res=0.001
)

print(U238.get_diagnostic_data(key='U238', data='synth_flu
```x'))

dvezinet, 2023-09-26:
Truns out plotting the two apertures and the detector in 3 shows there is indeed no available LOS through all of them:
[image]

[image]

shonmackie, 2023-09-27:
Fixing the LOS with correct nin_det vector resolved this issue