ToFuProject/tofu #952: Tutorial for collimators with multiple apertures

dvezinet, 2024-07-22:


jlball, 2024-07-22:
Error message for aperture list only:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jlball/NCAM/tomography/ncam_tomography/builders.py", line 256, in build_NCAM
    coll.add_diagnostic(
  File "/home/jlball/.local/lib/python3.8/site-packages/tofu/data/_class08_Diagnostic.py", line 90, in add_diagnostic
    dref, ddata, dobj = _check._diagnostics(
  File "/home/jlball/.local/lib/python3.8/site-packages/tofu/data/_class8_check.py", line 602, in _diagnostics
    ) = _diagnostics_check(
  File "/home/jlball/.local/lib/python3.8/site-packages/tofu/data/_class8_check.py", line 40, in _diagnostics_check
    doptics, lap, lfilt, lcryst, lgrat = _check_doptics_basics(
  File "/home/jlball/.local/lib/python3.8/site-packages/tofu/data/_class8_check.py", line 208, in _check_doptics_basics
    _err_doptics(key=key, doptics=doptics)
  File "/home/jlball/.local/lib/python3.8/site-packages/tofu/data/_class8_check.py", line 387, in _err_doptics
    raise Exception(msg)
Exception: diag 'ncam': arg doptics must be a dict with:
        - keys: key to existing camera
        - values: existing optics (aperture, filter, crystal)
                - as a list of str for regular cameras
                - as a list of (tuples of str) for collimator cameras

Provided:
{'cam0': ('cam0_ap00', 'cam0_ap01', 'cam0_ap10', 'cam0_ap11', 'cam0_ap20', 'cam0_ap21', 'cam0_ap30', 'cam0_ap31', 'cam0_ap40', 'cam0_ap41', 'cam0_ap50', 'cam0_ap51', 'cam0_ap60', 'cam0_ap61', 'cam0_ap70', 'cam0_ap71', 'cam0_ap80', 'cam0_ap81', 'cam0_ap90', 'cam0_ap91', 'cam0_ap100', 'cam0_ap101', 'cam0_ap110', 'cam0_ap111', 'cam0_ap120', 'cam0_ap121', 'cam0_ap130', 'cam0_ap131', 'cam0_ap140', 'cam0_ap141', 'cam0_ap150', 'cam0_ap151', 'cam0_ap160', 'cam0_ap161', 'cam0_ap170', 'cam0_ap171', 'cam0_ap180', 'cam0_ap181')}
```

Error message for aperture list and path matrix:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jlball/NCAM/tomography/ncam_tomography/builders.py", line 257, in build_NCAM
    coll.add_diagnostic(
  File "/home/jlball/.local/lib/python3.8/site-packages/tofu/data/_class08_Diagnostic.py", line 90, in add_diagnostic
    dref, ddata, dobj = _check._diagnostics(
  File "/home/jlball/.local/lib/python3.8/site-packages/tofu/data/_class8_check.py", line 602, in _diagnostics
    ) = _diagnostics_check(
  File "/home/jlball/.local/lib/python3.8/site-packages/tofu/data/_class8_check.py", line 40, in _diagnostics_check
    doptics, lap, lfilt, lcryst, lgrat = _check_doptics_basics(
  File "/home/jlball/.local/lib/python3.8/site-packages/tofu/data/_class8_check.py", line 208, in _check_doptics_basics
    _err_doptics(key=key, doptics=doptics)
  File "/home/jlball/.local/lib/python3.8/site-packages/tofu/data/_class8_check.py", line 387, in _err_doptics
    raise Exception(msg)
Exception: diag 'ncam': arg doptics must be a dict with:
        - keys: key to existing camera
        - values: existing optics (aperture, filter, crystal)
                - as a list of str for regular cameras
                - as a list of (tuples of str) for collimator cameras

Provided:
{'cam0': {'optics': ['cam0_ap00', 'cam0_ap01', 'cam0_ap20', 'cam0_ap21', 'cam0_ap40', 'cam0_ap41', 'cam0_ap60', 'cam0_ap61', 'cam0_ap80', 'cam0_ap81', 'cam0_ap100', 'cam0_ap101', 'cam0_ap120', 'cam0_ap121', 'cam0_ap140', 'cam0_ap141', 'cam0_ap160', 'cam0_ap161', 'cam0_ap180', 'cam0_ap181'], 'paths': array([[ True,  True, False, False, False, False, False, False, False,
        False, False, False, False, False, False, False, False, False,
        False, False],
       [False, False,  True,  True, False, False, False, False, False,
        False, False, False, False, False, False, False, False, False,
        False, False],
       [False, False, False, False,  True,  True, False, False, False,
        False, False, False, False, False, False, False, False, False,
        False, False],
       [False, False, False, False, False, False,  True,  True, False,
        False, False, False, False, False, False, False, False, False,
        False, False],
       [False, False, False, False, False, False, False, False,  True,
         True, False, False, False, False, False, False, False, False,
        False, False],
       [False, False, False, False, False, False, False, False, False,
        False,  True,  True, False, False, False, False, False, False,
        False, False],
       [False, False, False, False, False, False, False, False, False,
        False, False, False,  True,  True, False, False, False, False,
        False, False],
       [False, False, False, False, False, False, False, False, False,
        False, False, False, False, False,  True,  True, False, False,
        False, False],
       [False, False, False, False, False, False, False, False, False,
        False, False, False, False, False, False, False,  True,  True,
        False, False],
       [False, False, False, False, False, False, False, False, False,
        False, False, False, False, False, False, False, False, False,
         True,  True]])}}
```

jlball, 2024-07-22:
Code used to generate the camera and LOS:
Most of the activity is wrapped up into the `build_NCAM()` function:

```
# make sure to have an interactive backend
import sys
import os
import numpy as np
import matplotlib.pyplot as plt
from transp_profiles_2d import get_transp_profiles_2d
import bsplines2d as bs2d
from scipy.interpolate import RegularGridInterpolator
plt.ion()
#get_ipython().run_line_magic('matplotlib', 'qt5')
import tofu as tf
from pandas import read_csv

#######################################################
# Line of sight configurations
#######################################################

LOS_config_dict = {}

# All 19 lines of sight to be installed on SPARC
LOS_config_dict["all"] = [i for i in range(0, 19)]

# The 10 LOS currently being considered for NCAM
LOS_config_dict["10_los"] = [0, 2, 4, 6, 8, 10, 12, 14, 16, 18]

# 10 LOS NCAM geometry but with up / down asymmetry 
LOS_config_dict["10_los_asymmetric"] = [0, 2, 4, 6, 8, 9, 11, 13, 15, 17]
LOS_config_dict["7_los_asymmetric"] = [4, 6, 8, 9, 11, 13, 15]

LOS_config_dict["midplane_only"] = [7, 8, 9, 10, 11]

def build_NCAM(LOS_config="all", 
               transp_file="12345Q90.CDF", 
               geqdsk_file="SPARC_V1E_transp_3.geq", 
               config="SPARC-V0", 
               collimator_diameters=[1], 
               sensor_collimator_dist=10,
               sensor_side_length=3,
               sensor_shape = "square", 
               angle_tolerance=0,
               endpoint_tolerance=0):
    #######################################################
    # Configure Tofu
    #######################################################
    # simpler config => faster computation and plotting
    conf = tf.load_config(config)

    # instanciate a Diagnostic
    coll = tf.data.Collection()

    #Get emmissivity profile and its position grids
    r_grid = np.linspace(1.85 - 0.6, 1.85 + 0.6, num=100)
    z_grid = np.linspace(-0.6*1.8, 0.6*1.8, num=200)

    z_mesh, r_mesh = np.meshgrid(z_grid, r_grid)

    data_dict = get_transp_profiles_2d(5.973, f'data/{transp_file}', f'data/{geqdsk_file}', r_grid, z_grid)

    # Populate with a rectangular mesh
    coll.add_mesh_2d_rect(
        # naming
        key='m0',
        # parameters
        domain=None,           # if res is provided, assumes a uniform mesh, for rectangular use res = [resR, resZ]
        knots0 = r_grid,         # if res not provided, can provide the R vector of knots directly
        knots1 = z_grid,         # if res not provided, can provide the Z vector of knots directly
[... 148 lines cut ...]
        dgeom['cent'] = coll_end_cent + collimator_length * nin
        coll.add_aperture(key=kap1, **dict(dgeom))

        # fill in doptics
        #doptics["cam0"].append((kap1, kap0))
        # doptics["cam0"]["paths"][i, 2*i] = True
        # doptics["cam0"]["paths"][i, 2*i + 1] = True
        doptics["cam0"] += [kap0, kap1]
        
    # ---------------------------------------------------------------
    # create the diagnostic ncam1 using the info contained in doptics

    # transforming to tuple => collimator camera
    doptics['cam0'] = tuple(doptics['cam0'])
    #print(doptics)

    coll.add_diagnostic(
        key='ncam',
        doptics=doptics,
        compute=True,
        config=conf,
    )   

    return coll
```

dvezinet, 2024-07-22:
OK, I see where the issue is, it's not a bug in tofu,:

You're adding apertures and combining them into a diagnostic

But you forgot to add the camera too.

Indeed, the `doptics` dict should have a key that's refereing to an existing camera, but here `'cam0'` refers to nothing as it was never created.

I would suggest that before your loop on lines of sight, you initialize a `dgeom` dict that will hold the geometry of the camera (the camera being a collection of pixels).

Inside the loop, for each LOS, you append the content of dgeom to declare each pixel

Then , after the loop, but before you call `add_diagnostic()`, you call `add_camera_1d(key='cam0', dgeom=dgeom)` to add the camera so  `'cam0'` actually corresponds to an existing camera when tofu inspects the `doptics` dict.  

Although that did not correspond to a bug in tofu, i will use this feedback to improve the error message and make it more informative.
Let me know how it goes and if you need more help,