ToFuProject/tofu #999: Is wavelength bspline required when defining emissivity data?

cjperks7, 2024-12-13:
Maybe I'm setting up my emissivity data wrong, but I'm getting the following error when running `coll.compute_diagnostic_signal`:
[image]

The following code is stuff I ripping from `tofu_sparc` so things might have changed.

When I define my (R,Z) mesh and bsplines, I do:
[image]

When I define my (wavelength) mesh, I do:
[image]

When I define my (emissivity) mesh, I do:
[image]

When I define my wavelength mesh, I had commented out making a bspline since it wasn't needed, but I guess now it is? What I'm guessing I should do to fix things is
1) replace my commented out wavelength bspline with:
```
coll.add_bsplines(
            key='nlamb_'+key_diag,
            deg=1
            )
```
to match how I did it with the (R,Z) mesh and to maintain the reference

2) replace the reference in the emissivity data to `ref=('m0_bs1','nlamb_'+key_diag+'_bs1')`

Does that sound right? What do you think?

Didou09, 2024-12-13:
So yes, I remember I had started re-working that part and may have made spectral bsplines necessary without noticing.

I'd advise you basically un-comment the spectral mesh creation, and instead comment out the add_ref() and add_data().

```
coll.add_mesh_1d(
     key='mlamb_'+key_diag,
     knots=lamb*1e-10,
     deg=1,
    units='m',
)
```

Then just declare the spectral mesh as a reference for the emissivity
```
coll.add_data(
    key='emiss_'+key_diag,
    data=emiss2d_tmp_m0,
    ref=('m0_bs2', 'mlamb_bs1'),
    units='ph/s/m3/sr/m',
)
```

cjperks7, 2024-12-13:
Sounds good! I'll test a low-resolution version and let you know if it works out

cjperks7, 2024-12-13:
Seems like things are working now. Thank you very much!

closed 2024-12-13 by cjperks7