ToFuProject/tofu #1016: imas2tofu with ids='core_profiles'

tbarbui, 2025-01-24:
```
multi = tf.imas2tofu.MultiIDSLoader(shot=60763, ids='core_profiles')

plasma = multi.to_Plasma2D(strict=False)
```

```
core_profiles	ip data	empty data in core_profiles.ip
/Home/TB275660/Documents/Tofu/tofu/tofu/imas2tofu/_comp.py:869: UserWarning: The following data could not be retrieved:
	- core_profiles:
		1dTi_av data:  empty data in core_profiles.1dTi_av
		1dnW data   :  'profiles_1d_ion__structArrayElement' object has no attribute 'identifier'
		ip data     :  empty data in core_profiles.ip
		vloop data  :  empty data in core_profiles.vloop
  warnings.warn(msg)
Traceback (most recent call last):

  File "/tmp/ipykernel_42273/993560339.py", line 1, in <cell line: 1>
    plasma = multi.to_Plasma2D(strict=False)

  File "/Home/TB275660/Documents/Tofu/tofu/tofu/imas2tofu/_core.py", line 2047, in to_Plasma2D
    d0d, dtime0 = self._get_dextra(dextra)

  File "/Home/TB275660/Documents/Tofu/tofu/tofu/imas2tofu/_core.py", line 1885, in _get_dextra
    out = self.get_data(

  File "/Home/TB275660/Documents/Tofu/tofu/tofu/imas2tofu/_core.py", line 1596, in get_data
    return _comp.get_data_units(

  File "/Home/TB275660/Documents/Tofu/tofu/tofu/imas2tofu/_comp.py", line 853, in get_data_units
    raise verr

Exception: empty data in `core_profiles.ip`

```

--> This error is solved by modifying the following lines in /imas2tofu/_core.py:

 line 1841 as : `def _get_dextra(self, dextra=None, fordata=False, nan=True, pos=None, stack=None, strict=None):`

 line 1885 as: `out = self.get_data(dsig={ids: vs}, nan=nan, pos=pos, stack=stack, strict=strict, return_all=False)[ids]`

 line 2047 as: `d0d, dtime0 = self._get_dextra(dextra, strict=strict)`

tbarbui, 2025-01-24:
However another Exception is raised afterwards

```
plasma = multi.to_Plasma2D(strict=False)

/Home/TB275660/Documents/Tofu/tofu/tofu/imas2tofu/_comp.py:869: UserWarning: The following data could not be retrieved:
	- core_profiles:
		1dTi_av data:  empty data in core_profiles.1dTi_av
		1dnW data   :  'profiles_1d_ion__structArrayElement' object has no attribute 'identifier'
		ip data     :  empty data in core_profiles.ip
		vloop data  :  empty data in core_profiles.vloop
  warnings.warn(msg)
/Home/TB275660/Documents/Tofu/tofu/tofu/imas2tofu/_comp.py:869: UserWarning: The following data could not be retrieved:
	- core_profiles:
		ip data   :  empty data in core_profiles.ip
		vloop data:  empty data in core_profiles.vloop
  warnings.warn(msg)
Traceback (most recent call last):

  File "/tmp/ipykernel_42273/993560339.py", line 1, in <cell line: 1>
    plasma = multi.to_Plasma2D(strict=False)

  File "/Home/TB275660/Documents/Tofu/tofu/tofu/imas2tofu/_core.py", line 2050, in to_Plasma2D
    return _comp_toobjects.get_plasma(

  File "/Home/TB275660/Documents/Tofu/tofu/tofu/imas2tofu/_comp_toobjects.py", line 728, in get_plasma
    raise Exception(msg)

Exception: No 2d radius for polar mesh!
```

tbarbui, 2025-01-24:
Restricting the loaded data to some shortcuts does not solve the problem

```
dsig = {'core_profiles': ['t', '1dne', '1dTe', '1drhotn']}

plasma = multi.to_Plasma2D(strict=False, dsig=dsig)

/Home/TB275660/Documents/Tofu/tofu/tofu/imas2tofu/_comp.py:869: UserWarning: The following data could not be retrieved:
	- core_profiles:
		ip data   :  empty data in core_profiles.ip
		vloop data:  empty data in core_profiles.vloop
  warnings.warn(msg)
Traceback (most recent call last):

  File "/tmp/ipykernel_31594/2674190622.py", line 1, in <cell line: 1>
    plasma = multi.to_Plasma2D(strict=False, dsig=dsig)

  File "/Home/TB275660/Documents/Tofu/tofu/tofu/imas2tofu/_core.py", line 2050, in to_Plasma2D
    return _comp_toobjects.get_plasma(

  File "/Home/TB275660/Documents/Tofu/tofu/tofu/imas2tofu/_comp_toobjects.py", line 683, in get_plasma
    raise Exception(msg)

Exception: No valid radial base could be identified!
A valid radial base should be a 1d monotonous array
```

Didou09, 2025-01-24:
Hum, @tbarbui , are you available on Sunday ? or Tuesday ?

tbarbui, 2025-01-27:
Hi Didier,

I am available on Tue before 3pm French time.
Thank you!

Best,
Tullio
--
Dr. Tullio Barbui

Staff Research Physicist
Princeton Plasma Physics Laboratory
On assignment at
IRFM - CEA Cadarache
13108 Saint-Paul-lez-Durance
France
Phone: +33(0)665537203

closed 2025-02-10 by dvezinet